balena-io/etcher #3198: Linux Lite ISO Burn fails with gui.js:122599 error Error: e

linuxlite, 2020-06-03:
- **Etcher version:** 1.5.95
- **Operating system and architecture:** Linux Lite 5.0 (Ubuntu 20.04 base)
- **Image flashed:** linux-lite-5.0-64bit.iso
- **Do you see any meaningful error information in the DevTools?** 

````
gui.js:79012  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.95, Type = deb
gui.js:129897 Wed Jun 03 2020 19:27:02 GMT+1200 (New Zealand Standard Time) Application start ({"packageType":"deb","version":"1.5.95","sample":0.1,"applicationSessionUuid":"0b2f4db8-3c56-405c-9a79-08f0979119ca","flashingWorkflowUuid":"37de58d6-9093-4cc9-bc85-8fc8d814139e"})
gui.js:122599 Wed Jun 03 2020 19:27:21 GMT+1200 (New Zealand Standard Time) Open image selector ({"sample":0.1,"applicationSessionUuid":"0b2f4db8-3c56-405c-9a79-08f0979119ca","flashingWorkflowUuid":"37de58d6-9093-4cc9-bc85-8fc8d814139e"})
gui.js:122599 Wed Jun 03 2020 19:27:23 GMT+1200 (New Zealand Standard Time) Select image ({"image":{"size":1405091840,"name":"linux-lite-5.0-64bit.iso","hasMBR":true,"partitions":[{"offset":176128,"size":3833856,"type":239,"index":1}],"path":"/home/jerry/linux-lite-5.0-64bit.iso","extension":"iso","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"0b2f4db8-3c56-405c-9a79-08f0979119ca","flashingWorkflowUuid":"37de58d6-9093-4cc9-bc85-8fc8d814139e"})
gui.js:122599 Wed Jun 03 2020 19:27:31 GMT+1200 (New Zealand Standard Time) Flash ({"image":"/home/jerry/linux-lite-5.0-64bit.iso","drives":[{"size":4026531840,"isVirtual":false,"enumerator":"lsblk:json","logicalBlockSize":512,"raw":"/dev/sdb","error":null,"isReadOnly":false,"displayName":"/dev/sdb","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/sdb","busVersion":null,"isSystem":false,"busType":"USB","isCard":null,"isUSB":true,"devicePath":"/dev/disk/by-path/pci-0000:00:14.0-usb-0:6:1.0-scsi-0:0:0:0","mountpoints":[],"description":"Linux Lite 5.0 IS917 USB_3.0","isUAS":null}],"driveCount":1,"uuid":"959d9903-35b4-4b3e-a9ca-8799eed2ecc6","status":"started","flashInstanceUuid":"959d9903-35b4-4b3e-a9ca-8799eed2ecc6","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1,"applicationSessionUuid":"0b2f4db8-3c56-405c-9a79-08f0979119ca","flashingWorkflowUuid":"37de58d6-9093-4cc9-bc85-8fc8d814139e"})
gui.js:122599 Elevating command: /opt/balenaEtcher/balena-etcher-electron.bin /opt/balenaEtcher/resources/app
gui.js:122599 Wed Jun 03 2020 19:27:31 GMT+1200 (New Zealand Standard Time) SafeWebview loaded ({"event":{"id":1,"url":"https://assets.balena.io/etcher-featured/index.html?etcher-version=1.5.95&api-version=2&optOutAnalytics=true","method":"GET","timestamp":1591169251663.479,"resourceType":"mainFrame","ip":"104.31.83.157","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["0"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["59d7b12cac0eeeaa-AKL"],"cf-request-id":["031aad0feb0000eeaa9d863200000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Wed, 03 Jun 2020 07:27:31 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Wed, 03 Jun 2020 06:18:36 GMT"],"last-modified":["Mon, 01 Jun 2020 19:09:20 GMT"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["1"],"x-fastly-request-id":["bcb37c420205eb747d462cb9519c883545cf11da"],"x-github-request-id":["8C32:1AF9:20EB24:2AC34F:5ED73E63"],"x-proxy-cache":["MISS"],"x-served-by":["cache-akl10324-AKL"],"x-timer":["S1591169251.362220,VS0,VE229"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"0b2f4db8-3c56-405c-9a79-08f0979119ca","flashingWorkflowUuid":"37de58d6-9093-4cc9-bc85-8fc8d814139e"})
gui.js:5613  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:122599 Successfully connected to IPC server: etcher-server-15757, socket root /run/user/1000/
gui.js:122599 Image: /home/jerry/linux-lite-5.0-64bit.iso
gui.js:122599 Devices: /dev/sdb
gui.js:122599 Umount on success: true
gui.js:122599 Validate on success: true
gui.js:122599 Auto blockmapping: true
gui.js:122599 Decompress first: true
gui.js:5613 Flashing 1 device, 1% at 20.90 MB/s (total 20.90 MB/s) eta in 66s with 0 failed devices
gui.js:5613 Flashing 1 device, 3% at 20.90 MB/s (total 20.90 MB/s) eta in 65s with 0 failed devices
gui.js:5613 Flashing 1 device, 5% at 20.90 MB/s (total 20.90 MB/s) eta in 64s with 0 failed devices
gui.js:5613 Flashing 1 device, 6% at 20.90 MB/s (total 20.90 MB/s) eta in 63s with 0 failed devices
gui.js:5613 Flashing 1 device, 7% at 20.90 MB/s (total 20.90 MB/s) eta in 62s with 0 failed devices
gui.js:5613 Flashing 1 device, 9% at 20.90 MB/s (total 20.90 MB/s) eta in 61s with 0 failed devices
gui.js:5613 Flashing 1 device, 11% at 20.90 MB/s (total 20.90 MB/s) eta in 60s with 0 failed devices
gui.js:5613 Flashing 1 device, 12% at 20.40 MB/s (total 20.40 MB/s) eta in 60s with 0 failed devices
gui.js:5613 Flashing 1 device, 14% at 20.40 MB/s (total 20.40 MB/s) eta in 59s with 0 failed devices
gui.js:5613 Flashing 1 device, 15% at 20.40 MB/s (total 20.40 MB/s) eta in 58s with 0 failed devices
gui.js:5613 Flashing 1 device, 17% at 21.00 MB/s (total 21.00 MB/s) eta in 55s with 0 failed devices
gui.js:5613 Flashing 1 device, 18% at 20.90 MB/s (total 20.90 MB/s) eta in 54s with 0 failed devices
gui.js:5613 Flashing 1 device, 20% at 20.90 MB/s (total 20.90 MB/s) eta in 53s with 0 failed devices
gui.js:5613 Flashing 1 device, 21% at 20.90 MB/s (total 20.90 MB/s) eta in 52s with 0 failed devices
gui.js:5613 Flashing 1 device, 23% at 20.90 MB/s (total 20.90 MB/s) eta in 51s with 0 failed devices
gui.js:5613 Flashing 1 device, 24% at 20.90 MB/s (total 20.90 MB/s) eta in 50s with 0 failed devices
gui.js:5613 Flashing 1 device, 26% at 21.00 MB/s (total 21.00 MB/s) eta in 49s with 0 failed devices
gui.js:5613 Flashing 1 device, 28% at 21.00 MB/s (total 21.00 MB/s) eta in 48s with 0 failed devices
gui.js:5613 Flashing 1 device, 29% at 20.40 MB/s (total 20.40 MB/s) eta in 48s with 0 failed devices
gui.js:5613 Flashing 1 device, 31% at 20.40 MB/s (total 20.40 MB/s) eta in 47s with 0 failed devices
gui.js:5613 Flashing 1 device, 32% at 20.90 MB/s (total 20.90 MB/s) eta in 45s with 0 failed devices
gui.js:5613 Flashing 1 device, 34% at 20.90 MB/s (total 20.90 MB/s) eta in 44s with 0 failed devices
gui.js:5613 Flashing 1 device, 35% at 20.90 MB/s (total 20.90 MB/s) eta in 43s with 0 failed devices
gui.js:5613 Flashing 1 device, 37% at 21.50 MB/s (total 21.50 MB/s) eta in 41s with 0 failed devices
gui.js:5613 Flashing 1 device, 39% at 20.90 MB/s (total 20.90 MB/s) eta in 41s with 0 failed devices
gui.js:5613 Flashing 1 device, 40% at 20.40 MB/s (total 20.40 MB/s) eta in 41s with 0 failed devices
gui.js:5613 Flashing 1 device, 42% at 20.90 MB/s (total 20.90 MB/s) eta in 39s with 0 failed devices
gui.js:5613 Flashing 1 device, 43% at 21.40 MB/s (total 21.40 MB/s) eta in 37s with 0 failed devices
gui.js:5613 Flashing 1 device, 45% at 21.50 MB/s (total 21.50 MB/s) eta in 36s with 0 failed devices
gui.js:5613 Flashing 1 device, 46% at 21.00 MB/s (total 21.00 MB/s) eta in 36s with 0 failed devices
gui.js:5613 Flashing 1 device, 48% at 21.00 MB/s (total 21.00 MB/s) eta in 35s with 0 failed devices
gui.js:5613 Flashing 1 device, 49% at 21.00 MB/s (total 21.00 MB/s) eta in 34s with 0 failed devices
gui.js:5613 Flashing 1 device, 51% at 20.40 MB/s (total 20.40 MB/s) eta in 33s with 0 failed devices
gui.js:5613 Flashing 1 device, 52% at 20.40 MB/s (total 20.40 MB/s) eta in 32s with 0 failed devices
gui.js:5613 Flashing 1 device, 54% at 20.90 MB/s (total 20.90 MB/s) eta in 31s with 0 failed devices
gui.js:5613 Flashing 1 device, 55% at 21.00 MB/s (total 21.00 MB/s) eta in 30s with 0 failed devices
gui.js:5613 Flashing 1 device, 57% at 21.00 MB/s (total 21.00 MB/s) eta in 29s with 0 failed devices
gui.js:5613 Flashing 1 device, 58% at 21.00 MB/s (total 21.00 MB/s) eta in 28s with 0 failed devices
gui.js:5613 Flashing 1 device, 60% at 21.00 MB/s (total 21.00 MB/s) eta in 27s with 0 failed devices
[... 50 lines cut ...]
async function (async)
flashImageToDrive @ gui.js:151283
tryFlash @ gui.js:151367
callback @ gui.js:151390
ba @ gui.js:99054
[... cut ...]
lc @ gui.js:99756
jc @ gui.js:99750
mc @ gui.js:99767
sc @ gui.js:99836
Ia @ gui.js:105066
Ma @ gui.js:99164
Rc @ gui.js:100102
id @ gui.js:100071
exports.unstable_runWithPriority @ gui.js:105395
cg @ gui.js:101263
Ga @ gui.js:105054
gd @ gui.js:100061
```

<!-- You can open DevTools by pressing `Ctrl+Shift+I` (`Ctrl+Alt+I` for Etcher before v1.3.x), or `Cmd+Opt+I` if you're on macOS. -->

![etcher](https://user-images.githubusercontent.com/5382578/83608411-017df000-a5d1-11ea-96d9-d41ad079872b.png)

linuxlite, 2020-06-03:
[image]

zvin, 2020-06-03:
Thanks for reporting this, I'll fix it @linuxlite .

linuxlite, 2020-06-03:
@zvin thank you. We are promoting Etcher a lot in the community.

linuxlite, 2020-06-09:
Fixed in 1.5.97 thank you.

closed 2020-06-09 by linuxlite