balena-io/etcher #3201: User needs more instructions regarding appImage for Linux x64

OdysLam, 2020-06-04:
Source: Internal Feedback
FD: https://www.flowdock.com/app/rulemotion/r-product/threads/zl402Xx97EEOMQfineWlKR6xsua

Issue: Mark downloaded a .zip file from etcher website but was unsure what to do with it. @konmouz mentioned that we should add additional instructions and re-assess the assumptions that we make regarding the user's knowledge.

davekimble2, 2020-06-12:
Or you could do that bit extra work and create a .deb file for Debian/Ubuntu/Mint/hundreds of other distros.

lurch, 2020-06-13:
> create a .deb file

???

https://github.com/balena-io/etcher/releases  :wink:

davekimble2, 2020-06-13:
OK, there is a .deb file on github, in a folder that has no link to it from the home page. 
But starting from https://www.balena.io/etcher it downloads a .zip file which unzips to an .appimage file.

On Ubuntu_20.04 downloading the .deb file and running Etcher_1.5.99 I get a message:
```
Attention

Drive USB_Storage (/dev/sdd) is unusually large for an SD-card or USB-stick.

Are you sure you want to flash this drive ?
```
Is 128 GB unusually large for a micro-SD card?  They go up to 1 TB these days.
Clicking on Continue, I get another message:
```
Attention

Something went wrong.  If it is a compressed image, please check that the archive is not corrupted.

No polkit authentication agent found.
```
What "polkit authentificaton agent" ?

lurch, 2020-06-13:
> Do you WANT people to use Etcher ?

Makes no difference to me. I'm not an Etcher developer, just a helpful community member. :man_shrugging: 
Perhaps you're looking for https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 ?

> What "polkit authentificaton agent" ?

Perhaps you're looking for https://github.com/balena-io/etcher/blob/master/FAQ.md#i-receive-no-polkit-authentication-agent-found-error-in-gnulinux ?