balena-io/etcher #3204: aim not install Debian Bullseye

carrabelloy, 2020-06-08:
Debian Bullseye läuft es nicht 

root@hp-workstation-carrabelloy-blog:/home/carrabelloy# sudo apt-get install balena-etcher-electron
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 balena-etcher-electron : Hängt ab von: libpango1.0-0 ist aber nicht installierbar
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.

lurch, 2020-06-09:
Duplicate of #3119 
This problem was fixed by #3140 but unfortunately the [bintray](https://bintray.com/etcher/debian) repos are **still** lagging behind. I believe it _should_ work if you download the latest `.deb` from https://github.com/balena-io/etcher/releases though?

@thundron / @zvin Any progress in resolving the bintray issues? I think there's been several issues in this repo which wouldn't even have cropped up in the first place if the version of Etcher hosted on Bintray was more up-to-date?

JOduMonT, 2020-07-01:
I have the same issue with the same OS

@lurch just tell us when it will be available

lurch, 2020-07-01:
@JOduMonT I have no idea, I don't work on Etcher! I'm just a helpful community member :slightly_smiling_face:

thundron, 2020-07-01:
Bintray repo should be on 1.5.99 now, which is just 1 behind.
We're going to change how we manage updates soon as we're not entirely happy with the current system.
1.5.99 should be high enough to fix the issue, closing.

closed 2020-07-01 by thundron

JOduMonT, 2020-07-02:
@thundron Where the 1.5.99 version come from
because apt search show me **1.5.76**
[image]
and on bintray the last version I see is **1.4.4**
https://dl.bintray.com/etcher/debian/

thundron, 2020-07-02:
@JOduMonT that's the old executable, now it's `balena-etcher-electron` and we *just* solved an issue with the releases information so you should be able to install/update to `balena-etcher-electron`

JOduMonT, 2020-07-02:
> now it's `balena-etcher-electron`

I presume you are talking about the bintray repo ?

thundron, 2020-07-02:
I'm talking about https://dl.bintray.com/etcher/debian/ <-- these are the old executables, we then renamed the application to balenaEtcher (balena-etcher for deb/rpm distribution).
Were you able to install v1.5.100 which got updated just now @JOduMonT?