balena-io/etcher #3205: Problems detecting and intialising CM3 on balenaFin

markcorbinuk, 2020-06-09:
- **Etcher version:** 1.5.96
- **Operating system and architecture:** Kubuntu 20.04 x86_64
- **Image flashed:** image_Raspbian-Fin-full.zip v0.1.0
- **Do you see any meaningful error information in the DevTools?** No

Etcher needs to be run twice to detect the Compute Module on the balenaFin (v1.0).
The first time Etcher runs it gets stuck initialising the device (as shown below):
[image]
If you quit and start Etcher again the Compute Module automatically appears as the target (as shown below):
[image]
At this point you can continue and program the device successfully.

Steps to reproduce are as follows:
1) Insert debug cable into balenaFin.
2) Power up balenaFin.
3) Run Etcher as root.
4) Select 'Flash from File'.
5) Select 'image_Raspbian-Fin-full.zip'.
6) The 'Select target' button is now available, so click it.
7) The 'Select a Drive' dialog shows 'Compute Module - Initializing device'. When the progress bar is almost complete (maybe 90%) it doesn't move any further.
8) If you exit from Etcher and immediately restart it the Compute Module automatically appears as the target.
9) You can then select 'Flash from file' and program the device.
10) Once the flashing process has completed the board can be booted into Raspbian.

thundron, 2020-06-10:
Interesting, I'll try once more on my ubuntu VM, I can see it being slow (requiring up to a few minutes depending on the system) but not stuck

markcorbinuk, 2020-06-10:
It seems that once it has reached the progress shown in the screenshot it is actually 'ready' because when you restart Etcher it is detected immediately. Up until now I've left it to run for about 15 minutes before giving up. I'll let it run longer and see if it completes. Let me know if there is anything I can do to help debug this.

markcorbinuk, 2020-06-10:
It's been running for 2.5 hours this morning and is still stuck.

thundron, 2020-06-11:
I shall correct myself - I wasn't able to reproduce this specific issue anymore, but there are multiple instances of the compute module appearing as "initialising" when just one is connected

batesenergy, 2021-08-17:
Anything more here? Our compute module is never found when using with Ubuntu?

batesenergy, 2021-08-17:
Hi, We are having a similiar issue:

Linux Desktop w/ Ubuntu 21.04
Balena Fin with CM3 Lite Module

Balena Etcher run from appImage won't find compute module no matter how many times i plug and unplug it, etc. Running lsblk does report back a broadcom device connected? Any ideas?

thundron, 2021-08-20:
@batesenergy you should run it with `sudo` for usbboot to work 

also @markcorbinuk does this still happen? I think we solved this a long time ago

markcorbinuk, 2021-08-20:
@thundron I haven't tried this for a long time. The original problem was related to USB timing. It only occurred when using my laptop (my build machine worked fine). Increasing the `await delay` in `node-raspberrypi-usbboot` solved the problem for me.

batesenergy, 2021-08-23:
@markcorbinuk Is this something I can tweak on my local machine?

markcorbinuk, 2021-08-23:
@batesenergy This isn't my area of expertise - maybe @thundron can provide some guidance here?