balena-io/etcher #3213: Download for Linux 32bit/64bit stuck at 1.5.45

doczkal, 2020-06-13:
For me the download from https://www.balena.io/etcher/ shows 1.5.45 for the AppImage zip-file while github has release 1.5.99.

1.5.45 is ~1 year old. Who can update the web page and point to the latest version?

lurch, 2020-06-13:
IIRC the Etcher website sometimes shows old versions if you have JavaScript turned off? :shrug:

doczkal, 2020-06-14:
Who is able to modify the default if JavaScript is blocked?  

I can confirm that:
- while blocking scripts from  raw.githuberusercontent.com is see 1.5.45
- while granting scripts from raw.githubusercontent.com I see 1.5.99

oliverhbailey, 2020-06-15:
Etcher was once a very nice product. Now if the SD card has ever been used, even if the partitions are unallocated, the card can't be used. you should be able to use a card with no used or allocated partitions. Disappointing product!

lurch, 2020-06-15:
@oliverhbailey If you have a problem, you should probably create a new issue, rather than commenting on an unrelated issue? :man_shrugging:

oliverhbailey, 2020-06-15:
It doesn't appear to be a problem, it appears to be a shortcoming. There is
no indication it is acting abnormally!!

Thank you,
Oliver Bailey

doczkal, 2020-06-19:
@oliverhbailey please do not hijack my issue but create your own. Thanks!

oliverhbailey, 2020-06-19:
I have no idea what you're talking about. And frankly, IDGAF. I was posting a comment and nothing more. And that's exactly what it was, a comment. I didn't hijsck anything. I can use dd with a lot less hassle and did.

lurch, 2020-06-20:
The place for "comments" is probably https://forums.balena.io/c/etcher rather than GitHub issues (which is for reporting bugs / suggesting new features) ?
All IMHO. (I don't work on Etcher, I'm just a helpful community member)

thundron, 2020-06-25:
I think it's fair to ask to create a new issue if the problem doesn't fit the current one and most importantly *behave correctly*.
returning to @doczkal - you might need to enable javascript on the website, that's the fallback version in case your browser blocks our request to get the latest version

closed 2020-06-25 by thundron