balena-io/etcher #3215: [Windows 10 x64 1903]Writer process ended unexpectedly

unm4sk1g, 2020-06-17:
- **Etcher version:** 1.5.99+5d95fcb8, Type = nsis
- **Operating system and architecture:** Windows 10 x64 1903
- **Image flashed:** ubuntu-20.04-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?** 
```
"error":{"description":"Please try again, and contact the Etcher team if the problem persists","report":false,"code":"ECHILDDIED"},"sample":0.1,"applicationSessionUuid":"f0e458f3-3043-4e9a-be34-b3d3996cd507","flashingWorkflowUuid":"d0126961-8a80-4098-8442-7f6e86acae6b"})
```
**Steps I did**
- Followed [CONTRIBUTING.md](https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md) to build Etcher locally on my dev environment, then attempted to flash the provided image onto a 16 GB USB.

Following error occurred after elevating the cmd script:
[image]

If needed, I can provide more info about steps I took while building.

lurch, 2020-06-17:
Out of curiosity, do you also get the same error in the "official" v1.5.99 release downloaded from https://github.com/balena-io/etcher/releases or is this error specific to your local build of Etcher?

unm4sk1g, 2020-06-17:
Official 1.5.99 release works as intended, but I'm trying to figure out why locally built version without any code changes fails. I guess I'm missing something out here.

thundron, 2020-06-17:
> If needed, I can provide more info about steps I took while building.

@unm4sk1g it would be super-useful if you could do that.
Also: are you running etcher as a built package or through `npm start`?

unm4sk1g, 2020-06-17:
1. Installed fresh installation of Windows 10 Pro x64 1903, along with dependencies required in contributing document (NSIS 2.51, Python2.7, VS with Desktop C++ env, jq, msys-make etc.)
2. Cloned the repository recursively with scripts submodule.
3. Ran `make electron-develop`
Had some issues here, some dependencies required mingw32-make, for some reason things like shasum and sha1sum were missing, managed to fix that.
4. Finally managed to install all dependencies and ran `make webpack`
5. After everything, ran `make electron-build` and used the output setup to install Etcher on my PC.
6. Tried to flash an image, got thrown the error above.

And yes, I'm running etcher as a built package.

unm4sk1g, 2020-06-22:
Any way to fix this, I'm running out of ideas?

unm4sk1g, 2020-06-23:
I managed to narrow a conclusion that this only happens on Linux based images, Windows image flashed fine, even though it's missing the bootloader, it managed to flash successfully. Linux-based images (Fedora, Ubuntu etc.) throw the error instantly.

thundron, 2020-06-23:
we need to investigate a bit further. do you remember which packages gave you trouble?

unm4sk1g, 2020-06-23:
I'm about to re-do the whole process of building and pass the results along here.

unm4sk1g, 2020-06-23:
`make electron-develop` log: https://pastebin.com/QBwAFwpQ
`make electron-build` log: https://pastebin.com/7WAs8jDs
full chrome devtools log: https://pastebin.com/MFZMVtyx

I noticed this though when i ran `make electron-test`: https://pastebin.com/AA4uMmAx

maybe this is causing issues?

thundron, 2020-06-23:
@unm4sk1g it could be, that's the module we use to do direct-io operations (read, write, etc)

unm4sk1g, 2020-06-23:
But the module didn't throw any errors on node-gyp rebuild, as you can see in develop logs, so I'm curious what's causing it to miscompile.

thundron, 2020-06-23:
@unm4sk1g no, but the clean-shrinkwrap script fails. If you have any changes in the shrinkwrap file, you should commit them before running `make electron-develop` again, even better if you commit the changes (e.g. to a local branch) and then remove the node_modules folder so that the modules are installed from a clean state

unm4sk1g, 2020-06-24:
I have now pulled the latest changes from `master`, overwriting the shrinkwrap file with the one from `master`. `make electron-develop` passed correctly, `make electron-test` ran as well without any problems, so no more `direct-io` binding win32 error. However, issue is still the same. ECHILDDIED on Linux images.

unm4sk1g, 2020-07-01:
7 days have passed, re-created the whole environment and re-did all the steps from CONTRIBUTING.md, issue still persists. I'm out of ideas from here.

thundron, 2020-07-01:
@unm4sk1g can you update `etcher-sdk` to `v4.1.17` on your branch then retry?
There might be some issues that we solved in the latest version which will be available on master soon

unm4sk1g, 2020-07-06:
@thundron I've updated etcher-sdk to v4.1.17, issue still persists.

I've added some console.logs to [child-writer.ts](https://github.com/balena-io/etcher/blob/master/lib/gui/modules/child-writer.ts), specifically on `ipc write` event just to see if it even comes to this event before it crashes.
Then I've executed the temporary cmd file manually from an elevated cmd before allowing it elevated permissions through UAC, just to see if there will be any output from the script itself.

This is the output:
```
C:\Users\sunje\AppData\Local\Temp>balena-etcher-electron--2456-n2CLDPWDwPe2-.cmd

C:\Users\sunje\AppData\Local\Temp>chcp 65001
Active code page: 65001

C:\Users\sunje\AppData\Local\Temp>set "IPC_SERVER_ID=etcher-server-2456"

C:\Users\sunje\AppData\Local\Temp>set "IPC_CLIENT_ID=etcher-client-2456"

C:\Users\sunje\AppData\Local\Temp>set "IPC_SOCKET_ROOT=C:\Users\sunje\AppData\Local\Temp\"

C:\Users\sunje\AppData\Local\Temp>set "ELECTRON_RUN_AS_NODE=1"

C:\Users\sunje\AppData\Local\Temp>set "UV_THREADPOOL_SIZE=64"

C:\Users\sunje\AppData\Local\Temp>set "SKIP=1"

C:\Users\sunje\AppData\Local\Temp>"C:\Users\sunje\AppData\Local\Temp\1eJu86M6u6uCJCmBnzf0pO3Tkaz\balenaEtcher.exe" "C:\Users\sunje\AppData\Local\Temp\1eJu86M6u6uCJCmBnzf0pO3Tkaz\resources\app"
WRITE STARTED

C:\Users\sunje\AppData\Local\Temp>
```

Only the console.log I've added is outputted from the script, nothing else. Don't know if that's normal behaviour or not, but the error still comes out the same as from the original issue above, terminate code: **ECHILDDIED**

closed 2021-08-12 by unm4sk1g