balena-io/etcher #3218: Verbiage Suggestion for Failed Flahs

RickCarlino, 2020-06-17:
- **Etcher version:** 1.5.99
- **Operating system and architecture:** Windows, Linux
- **Image flashed:** Not applicable.
- **Do you see any meaningful error information in the DevTools?** Not applicable, this is a UX issue.

(this is a cross-post from the forum- was not sure which channel was more appropriate)

Hi Folks!

First off, thanks for publishing Etcher. We currently use it for on boarding new [FarmBot](https://farm.bot) customers that need to flash the initial firmware on their SD card and it’s been great for that purpose.

One area that we think could be improved is the Flash Complete! message. Some less technical users have sent us support requests not understanding why their device will not boot.

When we ask to see a screenshot, we get this:
[image]

The large heading and the word “complete!” catches the eye of the user, but they fail to see the smaller failure message below that heading. I think that if a flash fails (even if just for one device), the word “failed” should be the most prominent and eye catching word on the screen.

This might not seem like a big deal to a technical user, but it’s worth mentioning that this has happened multiple times now. Typically the users that are affected by this are less technical than the average user of Etcher.

lurch, 2020-06-17:
See also #2894

rotorrad, 2020-06-20:
Mine keeps crashing too. Last time it happens three times in the row. restarted mac and the same thing again.