balena-io/etcher #3221: It's not Ecthing

Robertpuggle, 2020-06-21:
- **Etcher version:latest
- **Image flashed:recalbox-rpi3.img

my SD card for my Raspberry Pi 3 the exe didn't etched my img In the SD card and the card was 16G

limegreendragon, 2020-06-24:
It is easier to use terminal to write it to the SD card

First unmount the SD card:

_diskutil unmountDisk /dev/disk2_

 (the SD card I used was described as disk 2 - can check using line: 
_diskutil list_
it will be the one referred to as an external disk)

Next find the file path for the img file and make a note of it

Run code

_sudo dd bs=1m if=file_path_to_go_here.img of=/dev/disk2; sync_

If on a linux machine will need to change bs=1M, if using a mac use 1m

Hope this helps!

limegreendragon, 2020-06-24:
Sorry forgot to add:

To eject SD card run:

_sudo diskutil eject /dev/rdiskN_

lurch, 2020-06-24:
> It is easier to use terminal to write it to the SD card

"Easier" for one person is "more difficult" for another person - the trickiness with using `dd` was exactly why Etcher was developed in the first place :wink:

thundron, 2020-06-25:
@Robertpuggle could you be a bit more specific about the version number and the error message you saw + at what point of the process you saw it?