balena-io/etcher #3225: xxhash requires un-fitted NODE_MODULE_VERSION

awesometic, 2020-06-26:
- **Etcher version:** 1.5.100
- **Operating system and architecture:** macOS 10.15.5 and Ubuntu 20.04 LTS
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** Yes, please see the contents below.

Hi.

I want to build Etcher but I cannot on my side.

What I did are,

```bash
make distclean # To make sure it starts from the scratch
npm install
make electron-develop
npm start
```

The screen shows nothing but a gray screen without any widget.

[image]

The Developer tools say,

[image]

Then I could get the below error when I enter `npm test`.

```bash
~/Developer/etcher-test master* ❯ npm test                                                                                                                                                                 49s 12:24:58

> balena-etcher@1.5.100 test /Users/awesometic/Developer/etcher-test
> make lint test sanity-checks

balena-lint --fix --typescript typings lib tests scripts/clean-shrinkwrap.ts webpack.config.ts

sass-lint -v lib/gui/app/scss/**/*.scss lib/gui/app/scss/*.scss
cpplint --recursive src
Done processing src/utils/v8utils.cpp
Done processing src/os/elevate.h
Done processing src/elevator_init.cpp
Done processing src/os/win32/elevate.cpp
Done processing src/utils/v8utils.h
codespell \
                --dictionary - \
                --dictionary dictionary.txt \
                --skip *.svg *.gz,*.bz2,*.xz,*.zip,*.img,*.dmg,*.iso,*.rpi-sdcard,*.wic,.DS_Store,*.dtb,*.dtbo,*.dat,*.elf,*.bin,*.foo,xz-without-extension \
                lib tests docs Makefile *.md LICENSE
WARNING: Binary file: lib/gui/app/scss/fonts/SourceSansPro-Regular.ttf
WARNING: Binary file: lib/gui/app/scss/fonts/SourceSansPro-SemiBold.ttf
WARNING: Binary file: tests/data/wmic-output.txt
electron-mocha --recursive --reporter spec --require ts-node/register --require-main "tests/gui/allow-renderer-process-reuse.ts" --full-trace --no-sandbox --renderer tests/gui/**/*.ts

✖ ERROR: The module '/Users/awesometic/Developer/etcher-test/node_modules/xxhash/build/Release/hash.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 80. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
Error: The module '/Users/awesometic/Developer/etcher-test/node_modules/xxhash/build/Release/hash.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 80. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:738:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
[... 68 lines cut ...]
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:738:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at /Users/awesometic/Developer/etcher-test/node_modules/electron-mocha/node_modules/mocha/lib/mocha.js:349:36
    at Array.forEach (<anonymous>)
    at Mocha.loadFiles (/Users/awesometic/Developer/etcher-test/node_modules/electron-mocha/node_modules/mocha/lib/mocha.js:346:14)
    at Mocha.run (/Users/awesometic/Developer/etcher-test/node_modules/electron-mocha/node_modules/mocha/lib/mocha.js:1006:10)
    at Object.runMocha (/Users/awesometic/Developer/etcher-test/node_modules/electron-mocha/lib/mocha.js:31:11)
    at EventEmitter.<anonymous> (/Users/awesometic/Developer/etcher-test/node_modules/electron-mocha/renderer/run.js:39:15)
    at EventEmitter.emit (events.js:223:5)
    at Object.onMessage (electron/js2c/renderer_init.js:2093:16)
(node:87249) Electron: Loading non-context-aware native module in renderer: '/Users/awesometic/Developer/etcher-test/node_modules/xxhash/build/Release/hash.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
make: *** [test-gui] Error 1
npm ERR! Test failed.  See above for more details.
```

I have tried with Node.js 8, 13, and 14 to change the NODE_MODULE_VERSION to 80, but doesn't work.
Also tried using `npm rebuild` and `npm install --save-dev electron-rebuild`, `./node_modules/.bin/electron-rebuild`, but all couldn't fix that issue.

It also doesn't work on Ubuntu 20.04 LTS either.

How can I build it on myself?

Thanks in advance.

lurch, 2020-06-26:
Have you looked at https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md ?

thundron, 2020-06-26:
@awesometic you should use just `make electron-develop` without `npm install`, try that instead and see how it goes

awesometic, 2020-06-26:
@lurch , Yes I have but it doesn't work as that document says. 

@thundron , You're right! I did `make distclean` then `make electron-develop`, then I can get it works! Thanks a lot. I'll close this issue.

closed 2020-06-26 by awesometic

lurch, 2020-06-26:
The document I linked to says:
> **NOTE:** Please make use of the following command to install npm dependencies rather than simply running `npm install` 
...
`make electron-develop`

:wink: :shrug: :upside_down_face:  Glad you've got it working now though.

awesometic, 2020-06-27:
@lurch , Oh actually I was able to build using that `npm install` before `make electron-develop` when I was trying to with 1.5.86 ? version. Maybe that experience makes me don't read the document anymore.. 😅 Sorry.