balena-io/etcher #3226: Filesize incorrectly reported for some bz2 images

richbayliss, 2020-06-26:
From the forums: https://forums.balena.io/t/144602

Using file OPNsense-20.1-OpenSSL-vga-amd64.img.bz2 (and others)
from site http://mirror.as24220.net/opnsense/releases/mirror/ (opnsense official Australian mirror)

Balena Etcher shows them as 9.44 ZB large. Understandably while writing I didn't wait for 1%....
If I use 7zip to decompress first, then write, image is 1.2 GB or so and writes just fine.

A bug in the bz2 decompress routine to do with the version of bz2 used on the opnsense files? No idea, just throwing ideas around. Anyway, I love the software, so had to share the hiccup.

OS: Win10 x64
Etcher v1.5.100

lurch, 2020-06-26:
AFAIK Etcher only "guesses" the size of a bz2 file (see e.g. https://superuser.com/questions/53984/is-there-a-way-to-determine-the-decompressed-size-of-a-bz2-file for the reason why), so it looks like this guess is going a bit wonky in some cases!

zvin, 2020-07-09:
This should be fixed in 1.5.101

closed 2020-07-09 by zvin