balena-io/etcher #3228: Writer process ended unexpectedly

TheRakeshPurohit, 2020-06-28:
- **Etcher version: 1.5.100** 
- **Operating system and architecture: Windows 10 2004 64Bit (OS Build: 19041.329)** 
- **Image flashed: POP OS 20.04 ** 
- **Do you see any meaningful error information in the DevTools?**

trolley, 2020-07-02:
I am getting this error with the same version of Etcher on macOS 10.15.5 trying to flash clonezilla-live-20200428-focal-amd64.iso.

jrhodk, 2020-07-02:
Im also getting this error in windows 10 and i have tried uninstalling and reinstalling a newly downloaded balenaetcher, to no avali. And I have tried countless different usbdrives and even a SD card, and different iso and img files. From Debian to Home Assistant.

Ntemis, 2020-07-04:
Same issue on ubuntu 20.04(focal)

TheRakeshPurohit, 2020-07-14:
Has anyone tried it on latest release 1.5.101 ?

agucova, 2020-07-14:
Same with Windows 10 freshly installed using 1.5.101. I tried elevating to admin privileges, no luck.

agucova, 2020-07-14:
The drive had been previously flashed by Rufus. I noticed the USB drive didn't show on the file manager and I used Rufus to flash "non-bootable" to the pendrive. Afterwards Etcher worked perfectly.