balena-io/etcher #3231: After the computer had gone to sleep, the software went blank.

CWZ7605, 2020-07-01:
- **Etcher version:All (1.5.100)**
- **Operating system and architecture:Windows 10 1909 (Build 18363.836)**
- **Image flashed: No** 
- **Do you see any meaningful error information in the DevTools? No** 

[image]
No new log information is generated in the DevTools.

After opening the software, I put the computer to sleep.When I wake up the computer again, the software goes blank.

curcuz, 2021-06-21:
I just got a similar issue on win10 where I left etcher open for a couple of hours (computer never went to sleep) and found it blank with console detached

[image]
[image]

thundron, 2021-07-07:
@CWZ7605 does this happen regularly? if so, could you please try to enable log persistence in the console and then try to reproduce the issue again? hopefully it will tell us what happened in the previous session even if the devtools disconnect

CWZ7605, 2021-07-12:
> @CWZ7605 does this happen regularly? if so, could you please try to enable log persistence in the console and then try to reproduce the issue again? hopefully it will tell us what happened in the previous session even if the devtools disconnect

Do you have detailed procedures? If so, I can follow it

thundron, 2021-07-13:
@CWZ7605 open the devtools as you did when it crashed, but do it *before* when the application is still running fine - go to the console tab and you should see in the top bar on the right a cog icon (the bottom one, not the top one) and then "preserve logs"

jonasjberg, 2025-02-14:
M1 laptop went to sleep during flashing.
When I noticed and woke it up the balanaEtcher application is wholly unresponsive.
MacOS 15.3 (24D60) but timeouts on your end should not be OS-specific? :)

None of the logs indicate any errors, just stops responding. Last log message was:
```
Flashing 1 device, 30% at 10.00 MB/s (total 10.00 MB/s) eta in 285s with 0 failed devices
```

I waited around for 30 minutes, then force-killed balanaEtcher. USB-stick required re-formatting Disk Utility to be usable again.
Second attempt was successful. So no biggie but this a serious bug surely.

Cheers :)