balena-io/etcher #3232: User would like ability to create an image from a storage device

saintaardvark, 2020-07-02:
As requested in the forums, user would like the ability to create a disk image from a storage device and burn that.  This would match the behaviour of Win32DiskImager.

lurch, 2020-07-03:
Duplicate of #266

thundron, 2020-07-07:
Correct, that's basically the backup function which we're working on (we should be one step closer with the clone drive feature, with which we'll be able to select a drive as a source).
Closing as duplicate

closed 2020-07-07 by thundron

ReanimationXP, 2026-09-22:
Why was this closed as completed when it's not complete? You guys implemented a clone feature as mentioned but not a backup to image file feature as requested. I think this would be very beneficial for the communities Etcher helps and would make it a one-stop tool for backing up a version of an SDcard and restoring it. This is critical for folks with a limited number of SDcards who want to temporarily use one for a different purpose. This issue is [referenced in various places](https://forums.balena.io/t/can-i-use-etcher-to-create-an-image-like-win32diskimager-does/3310/10) when the suggestion is brought up (but with a broken link) so I had to find it manually. I do not believe it to be a duplicate as cloning is a separate feature.