balena-io/etcher #3234: Etcher Issues with Windows Disk Management after imagining

PhilRib, 2020-07-05:
I have the latest version of etcher 1.5.100 which I downloaded to load the latest version of Sirius OS for a pi-top. Having initially loaded the OS without any issues I then needed to make another sd card version. This resulted in repeated failures with the error message 'The Writer Process Ended unexpectedly. Please try again and contact Etcher team if problem persists. The process of imagining was done on a windows 10 laptop and now I can no longer access Disk Management. I have tried the web based solutions without any success. Does anyone know how I can get my Sirius imaged for my Pi-Top and how I can return my laptop to where I can see the Disk Management tool

thundron, 2020-07-08:
@PhilRib for flashing the sd card, I suggest preemptively checking the Controlled Folder Access setting on Windows 10. In case that's turned on, add an exception for balenaEtcher so that it can access the resources.
Either way, we'd need the logs from the dev tools console to perform a better analysis so copy-paste them here whenever you encounter the issue.

With regards to the Disk Management tool, do you mean that you can't access the tool itself or just that you can't see the drive in it?

PhilRib, 2020-07-08:
Thanks for coming back to me on this I really appreciate it. I am not a particular expert in these things so you might have to excuse me if I have to ask a number of questions. I am currently running McAfee and so I have attempted to make an exemption balenaEtcher

The dev tools console I am afraid you will need to let me know what this is so I can access the details for you

 

Disk Management Tool comes up with an error – Unable Able to Connect to Virtual Disk Service – It was working fine as I checked the sd cards before running the Etcher program

 

Phil

thundron, 2020-07-08:
@PhilRib the dev tools console is accessible through pressing `CTRL+SHIFT+i`, it reveals the logs happening as the application is running and that should have a bit more insight on what's happening when the issue arises.
Antiviruses might get in the way of Etcher too so you did right by making an exception there.

As for the Disk Management tool, the first thing that pops up looking for that error on Google is https://www.partitionwizard.com/partitionmagic/unable-to-connect-to-virtual-disk-service-008.html so I'd suggest pursuing that path, we do use Diskpart, for example, to check the targets in the flash process so we need those resources to work properly. It might be that Etcher revealed an underlying issue that was already present as all we do is use those tools with the drive as the target and never to change any configuration

PhilRib, 2020-07-08:
OK – Thanks for your help – I have the DevTools working and have copied the entry from the Console area. Let me know if this is what you needed

 

Phil

 

gui.js:126777 Wed Jul 08 2020 14:46:44 GMT+0100 (British Summer Time) SafeWebview loaded ({"event":{"id":1,"url":"https://assets.balena.io/etcher-featured/index.html?etcher-version=1.5.100&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1594216004339.143,"resourceType":"mainFrame","ip":"104.31.82.157","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["389"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["5afa40d5ad860726-LHR"],"cf-request-id":["03d046d989000007260f17e200000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Wed, 08 Jul 2020 13:46:46 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Wed, 01 Jul 2020 06:25:27 GMT"],"last-modified":["Wed, 17 Jun 2020 21:19:41 GMT"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["1"],"x-fastly-request-id":["75a1ba2788834f0644fac6893fd87a93001599c3"],"x-github-request-id":["E768:33B3:2C4EC:36ACA:5EFC29FF"],"x-proxy-cache":["MISS"],"x-served-by":["cache-lcy19280-LCY"],"x-timer":["S1594216006.072005,VS0,VE0"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"adafb1f9-c145-4fba-a78f-c5fcda519f9d","flashingWorkflowUuid":"995c0ea7-5a55-43ce-9c97-952ced24e2b6"})

gui.js:126777 Successfully connected to IPC server: etcher-server-47640, socket root C:\Users\Phil Laptop\AppData\Local\Temp\

gui.js:126777 Image: C:\Users\Phil Laptop\Downloads\pi-topOS-sirius_C741_2020-05-18 (1).zip

gui.js:126777 Devices: \\.\PhysicalDrive3

gui.js:126777 Umount on success: true

gui.js:126777 Validate on success: true

gui.js:126777 Auto blockmapping: true

gui.js:126777 Decompress first: true

gui.js:126777 Wed Jul 08 2020 14:46:58 GMT+0100 (British Summer Time) Flash error ({"image":"C:\\Users\\Phil Laptop\\Downloads\\pi-topOS-sirius_C741_2020-05-18 (1).zip","drives":[{"size":32010928128,"isVirtual":false,"enumerator":"SCSI","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive3","error":null,"isReadOnly":false,"displayName":"K:\\","blockSize":512,"isSCSI":true,"isRemovable":true,"device":"\\\\.\\PhysicalDrive3","busVersion":"2.0","isSystem":false,"busType":"UNKNOWN","isCard":false,"isUSB":false,"devicePath":null,"mountpoints":[{"path":"K:\\"}],"description":"JMCR SD SCSI Disk Device","isUAS":false}],"driveCount":1,"uuid":"35aa9a5b-69a4-4eab-86fa-f5e383c98ce1","flashInstanceUuid":"35aa9a5b-69a4-4eab-86fa-f5e383c98ce1","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{"name":"Error","message":"Couldn't clean the drive, Command failed: diskpart s Phil Laptop\\AppData\\Local\\Temp\\etcher\\46dfafcac625.tmp\n (code 2147755009)","stack":"Error: Couldn't clean the drive, Command failed: diskpart s Phil Laptop\\AppData\\Local\\Temp\\etcher\\46dfafcac625.tmp\n (code 2147755009)\n at Object.exports.clean (C:\\Users\\Phil Laptop\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\etcher.js:54409:27)\n at async BlockDevice._open (C:\\Users\\Phil Laptop\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\etcher.js:20268:13)\n at async BlockDevice.open (C:\\Users\\Phil Laptop\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\etcher.js:7349:17)\n at async Phil Laptop\\AppData\\Local\\Programs\\balena-etcher\\resources\\app\\generated\\etcher.js:27361:21"},"sample":0.1,"applicationSessionUuid":"adafb1f9-c145-4fba-a78f-c5fcda519f9d","flashingWorkflowUuid":"995c0ea7-5a55-43ce-9c97-952ced24e2b6"})

gui.js:126777 Finish: 0

gui.js:126777 Terminating IPC server

gui.js:126777 Flash results 

1.      {results: {…}, cancelled: false}

gui.js:126777 Wed Jul 08 2020 14:46:58 GMT+0100 (British Summer Time) Write failed ({"image":"C:\\Users\\Phil Laptop\\Downloads\\pi-topOS-sirius_C741_2020-05-18 (1).zip","drives":[{"size":32010928128,"isVirtual":false,"enumerator":"SCSI","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive3","error":null,"isReadOnly":false,"displayName":"K:\\","blockSize":512,"isSCSI":true,"isRemovable":true,"device":"\\\\.\\PhysicalDrive3","busVersion":"2.0","isSystem":false,"busType":"UNKNOWN","isCard":false,"isUSB":false,"devicePath":null,"mountpoints":[{"path":"K:\\"}],"description":"JMCR SD SCSI Disk Device","isUAS":false}],"driveCount":1,"uuid":"35aa9a5b-69a4-4eab-86fa-f5e383c98ce1","status":"failed","flashInstanceUuid":"35aa9a5b-69a4-4eab-86fa-f5e383c98ce1","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{"description":"Please try again, and contact the Etcher team if the problem persists","report":false,"code":"ECHILDDIED"},"sample":0.1,"applicationSessionUuid":"adafb1f9-c145-4fba-a78f-c5fcda519f9d","flashingWorkflowUuid":"995c0ea7-5a55-43ce-9c97-952ced24e2b6"})

​

lurch, 2020-07-08:
@PhilRib Unrelated to this issue, but just a quick heads-up that you're replying to messages and including the entirety of the previous replies, which means that your email address is visible in plain-text for all the world to see at https://github.com/balena-io/etcher/issues/3234
:confused:

PhilRib, 2020-07-15:
@thundron and @lurch - Thanks very much for your comments - I looked at the Disk Management link provided and have tried the solutions suggested without success. I'll keep trying and see if I can find a solution

shawaj, 2020-12-13:
@PhilRib could you update to [v1.5.112](HTTPS://etcher.io) and try again?

Please also try running it as an administrator in Windows (right click on the icon and select run as administrator)

closed 2023-07-12 by dfunckt