balena-io/etcher #3235: Windows 10: Write fails but Windows popup notification says it was successful

POTSandSTOP, 2020-07-06:
Hi, I don't need help resolving the error, I'm just reporting the problem with the Windows notification popup.

- **Etcher version:** v1.5.100
- **Operating system and architecture:** Windows 10, AMD 64bit
- **Image flashed:** 2017-07-05-raspbian-jessie.zip
- **Do you see any meaningful error information in the DevTools?** 

[image]
[image]

lurch, 2020-07-06:
Duplicate of #2894 ?

POTSandSTOP, 2020-07-08:
I read #2894 and it is similar, except #2894 seems to be about "the message itself is not lying, since the writing (flash) process was a success" ([@KEINOS](https://github.com/KEINOS)). In my case, the writing did fail; I have a bad USB microSD card adapter.