balena-io/etcher #3238: Document recovery process for a USB drive in Windows

saintaardvark, 2020-07-09:
We get a lot of questions from Windows users about being unable to see USB drives after flashing them.  It would be good to take inspiration from https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive and provide an FAQ for this.

lurch, 2020-07-09:
See also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

dlsquires-safari, 2020-08-04:
I just came here to put in this feature request as well.  Since I don't want to add _another_
issue, I'll just try to explain my view of the situation.

# Description of problem

After writing a bootable image to a USB flash drive or SD card, it no longer can be utilized as
a traditional flash drive on PCs, instruments, or cameras.  Each of those devices expected a
storage drive to have one large partition formatted to FAT32 or exFAT.

The solution is to recovering a flash drive or SD card that is no longer needed as a bootable
drive is [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives).

# Expectations

Here is Balena Etcher's raison d'être

> Here at balena we have thousands of users working through our getting started process and
> until recently we were embarassed about the steps that involved flashing an SD card. There
> was a separate track for each Mac/Windows/Linux and several manual and error-prone steps
> along the way.
>
> To our surprise there was nothing out there that fit our needs. So we built Etcher, an SD card
> flasher app that is simple for end users, extensible for developers, and works on any platform.

The recovery process to get the USB flash driver or SD back into its original state is exactly as manual
and error prone as the original process Etcher was designed to fix.  This leaves a lot of people
who didn't have the technical experience to execute the original flashing process unable to restore
their USB flash drives and SD cards.  Frankly, if they can follow the directions given to restore those
drives, they would be able to follow the directions for the original processes of turning their drives
into bootable devices in the first place.

In short, it'd be great if Etcher had an easy option on all OS variant to clean and reformat drives
into the most common state that they come from the factory.

# References

Here are several related issues and forum posts

* #273
* #3032
* #3200
* #3191
* #3236
* #3268
* https://forums.balena.io/t/help-me-plz-my-external-hard-drive-dont-work-after-i-used-etcher/36928/2
* https://forums.balena.io/t/etcher-has-bricked-4-usb-sticks-and-3-sd-cards/27662/7

thundron, 2020-11-24:
we already have documentation at https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows on how to recover a drive + we have many issues open on the subject and a format fat32 on the pipeline, although we don't know when we'll be able to get to it.

PRs welcome as always, but closing this one as duplicate

closed 2020-11-24 by thundron