balena-io/etcher #3239: gui.js:125694 {"stack":"Error: The elevated process died unexpectedly\n    at createError (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:40976:23)\n    at Object.createUserError (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:40988:16)\n    at file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:156006:34\nFrom previous event:\n    at Object.elevateCommand (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:155985:25)\n    at async Server.<anonymous> (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:154963:37)","message":"The elevated process died unexpectedly","description":"The process error code was 126","report":false,"__stackCleaned__":true,"image":"CentOS-8.2.2004-x86_64-minimal.iso"}

ambidot, 2020-07-10:
- **Etcher version:** 1.5.101
- **Operating system and architecture:** Linux Mint 19.3
- **Image flashed:** CentOS-8.2.2004-x86_64-minimal.iso
- **Do you see any meaningful error information in the DevTools?** 

```
Something went wrong. If it is a compressed image, please check that the archive is not corrupted.

The elevated process died unexpectedly
```
- gui.js:125694 {"stack":"Error: The elevated process died unexpectedly\n    at createError (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:40976:23)\n    at Object.createUserError (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:40988:16)\n    at file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:156006:34\nFrom previous event:\n    at Object.elevateCommand (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:155985:25)\n    at async Server.<anonymous> (file:///tmp/.mount_balena0ZaliB/resources/app/generated/gui.js:154963:37)","message":"The elevated process died unexpectedly","description":"The process error code was 126","report":false,"__stackCleaned__":true,"image":"CentOS-8.2.2004-x86_64-minimal.iso"}

The iso doesn't appear to be corrupted; it downloaded without an error.

Alexandero89, 2020-07-17:
Please never put the complete exception stack as a title. Use a title which is human readable/understandable.
For Example: "The elevated process died unexpectedly"

alex27riva, 2020-07-29:
Same problem on Pop!_OS using the appimage format

jagin, 2020-08-20:
Same with Ubuntu 18.04 and AppImage

laspic, 2020-09-19:
Same with Ubuntu 18.04 and AppImage

jmkeol, 2020-10-04:
I have the same issue. It appears to be something related to the AppImage as other releases work fine for me.

simuuz, 2020-11-04:
It's probably a problem with the AppImage version, same problem here on Manjaro Linux (kernel 5.9.1-1)

cbstryker, 2021-02-09:
Is there no update on this? I've been having this issue for many months now and haven't been able to use Etcher at all.

Kubuntu 18.04

endorama, 2021-04-24:
I had this issue and solved following the suggestion in https://github.com/balena-io/etcher/issues/2881

TL;DR: remove `appimagelauncher` package, and eventually `libappimage`.

rvalle, 2022-03-26:
Same problem in Ubuntu appimage.... was working last week...

maks, 2023-02-12:
On ubuntu 22.04.1 I had this same issue and doing `sudo apt remove appimagelauncher` got balena working again straight away, not even a reboot required.

thediveo, 2024-01-08:
removing appimagelauncher is a no-go, because there's a reason for that tool. If balea is incompatible with it and it looks as if they don't give a fig in what 2 years??, then the only way for me is to basically warn our users to stick with other tools, that correctly work.