balena-io/etcher #3244: Ubuntu/Deb Repository Release file permissions

ThoreauHenry, 2020-07-12:
- **Etcher version:** 

Latest as of July 12, 2020.

- **Operating system and architecture:**
 
Ubuntu 20.04/Mint 20 x64

- **Image flashed:** 

Repo issue

- **Do you see any meaningful error information in the DevTools?** 

Repo Issue

Error:

E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.

N: Updating from such a repository can't be done securely, and is therefore disabled by default.

N: See apt-secure(8) manpage for repository creation and user configuration details.

>wget https://dl.bintray.com/etcher/debian/dists/etcher/Release.gpg

Forbidden!

>wget https://dl.bintray.com/etcher/debian/dists/etcher/Release

Forbidden!

lurch, 2020-07-12:
@thundron The Etcher bintray repo was working for me a couple of weeks ago, but now when I run `sudo apt update` on Ubuntu 18.04 it seems the bintray repo is broken again :cry: 
```
Ign:16 https://dl.bintray.com/etcher/debian stable InRelease                                      
Err:25 https://dl.bintray.com/etcher/debian stable Release
  403  Forbidden [IP: 35.157.101.235 443]
Reading package lists... Done                       
E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Looks like the files themselves are actually there https://dl.bintray.com/etcher/debian/dists/etcher/ but trying to view/download any file results in a 403 Forbidden error.

Eate, 2020-07-12:
Reproduced with Ubuntu 20.04 LTS amd64, PPA repository.

ndmax, 2020-07-12:
Repeat of #3078?

Eate, 2020-07-12:
> Repeat of #3078?

Looks like it.

hamie96, 2020-07-13:
I'm assuming this is problems with Bintray again.

InfoLibre, 2020-07-13:
Same problem o Linux Mint MATE 20.

```
$ sudo apt-get update
Ign :1 http://liveusb.info/multisystem/depot all InRelease
Atteint :2 http://mirror.23media.com/ubuntu focal InRelease                                                                                                                                                     
Atteint :3 http://mirror.23media.com/ubuntu focal-updates InRelease                                                                                                                                             
Atteint :4 http://liveusb.info/multisystem/depot all Release                                                                                                                                                            
Ign :5 http://mirror.netcologne.de/linuxmint/packages ulyana InRelease                                                                                                                                                  
Atteint :6 http://mirror.23media.com/ubuntu focal-backports InRelease                                                                                                                 
Atteint :7 http://mirror.netcologne.de/linuxmint/packages ulyana Release                                                                            
Atteint :9 http://archive.canonical.com/ubuntu focal InRelease                                                                                      
Réception de :10 http://security.ubuntu.com/ubuntu focal-security InRelease [107 kB]
Ign :8 https://dl.bintray.com/etcher/debian stable InRelease          
Err :11 https://dl.bintray.com/etcher/debian stable Release           
  403  Forbidden [IP : 3.120.60.234 443]
Lecture des listes de paquets... Fait
E: Le dépôt https://deb.etcher.io stable Release n'a pas de fichier Release.
N: Les mises à jour depuis un tel dépôt ne peuvent s'effectuer de manière sécurisée, et sont donc désactivées par défaut.
N: Voir les pages de manuel d'apt-secure(8) pour la création des dépôts et les détails de configuration d'un utilisateur.
```

CharlesCage, 2020-07-13:
Reproduced with Debian Buster.

thundron, 2020-07-13:
Thanks for the reports, we're looking into it

CristianH033, 2020-07-14:
Is this why some people think it's better to use flatpak or snap?

InfoLibre, 2020-07-14:
Sanp, Flatpack and AppImage are bad thinks. No automatic updates, full disks, nobody to verofy the contents.

YtvwlD, 2020-07-14:
Snap and Flatpak have automatic updates. And when downloading over HTTPS I don't think there's a need to verify the contents, but ymmv.

lurch, 2020-07-14:
Each packaging format has its own pros and cons, IMHO :wink:

laurent-simon, 2020-07-14:
> Is this why some people think it's better to use flatpak or snap?

No. It's a server configuration problem. You can have the same kind of problems with Flatpak or Snap.

The only case where it's better to use a solution like Flatpack or Snap is when you cannot do anything else than duplicating everything to resolve software dependencies problems. In this case, these solutions are an easy way to do it. In all other cases, they just bloat and slow down your system, without any interesting return.

GammaGames, 2020-07-14:
Possibly #3078 showing up again?

Off topic:

> And when downloading over HTTPS I don't think there's a need to verify the contents, but ymmv.

@YtvwlD That is not what they mean when they say verify contents. They are referring to the fact that an official APT repo will have tested the package contents and verified that there aren't any obvious exploits included. This probably isn't the case for a third party PPA like etcher, but the concept in general is valuable.

mmaalo, 2020-07-14:
> @thundron The Etcher bintray repo was working for me a couple of weeks ago, but now when I run `sudo apt update` on Ubuntu 18.04 it seems the bintray repo is broken again 
> 
> ```
> Ign:16 https://dl.bintray.com/etcher/debian stable InRelease                                      
> Err:25 https://dl.bintray.com/etcher/debian stable Release
>   403  Forbidden [IP: 35.157.101.235 443]
> Reading package lists... Done                       
> E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
> N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> N: See apt-secure(8) manpage for repository creation and user configuration details.
> ```
> 
> Looks like the files themselves are actually there https://dl.bintray.com/etcher/debian/dists/etcher/ but trying to view/download any file results in a 403 Forbidden error.

`sudo apt-get update
...
Feil:13 https://dl.bintray.com/etcher/debian stable Release
  403  Forbidden [IP: 35.157.101.235 443]
...
E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.
N: Oppdatering fra et slikt depot kan ikke gjøres på sikkert vis, og er derfor avskrudd som standard.
N: Se manualsiden apt-secure(8) for detaljer om oppretting av depote og brukeroppsett.
`

I get the same error on debian 10 (buster)

darkvertex, 2020-07-14:
Getting this problem also, on Ubuntu 18.04.4 LTS.

nebrelbug, 2020-07-15:
I'm having the same problem. Just a BTW, I'd love to have a Flatpak release. Have there been any updates on the technical roadblocks?

sha-265, 2020-07-16:
ACK

makemegit, 2020-07-17:
I am having the same problem from a couple of days ago on my ubuntu 18.04.

```
curl https://dl.bintray.com/etcher/debian/dists/etcher/Release
Forbidden!
```

jaylittle, 2020-07-17:
It's absurd that this hasn't been fixed yet.  I've decided to uninstall etcher entirely and begin using an alternative as I'm tired of my package manager screaming at me.

So long and thanks for all the fish Etcher devs!

sajanp, 2020-07-18:
@jaylittle just for sake of comparison, how much did you pay for your SLA with balena?

jaylittle, 2020-07-18:
> @jaylittle just for sake of comparison, how much did you pay for your SLA with balena?

Not a single penny. But having this PPA down prevents a lot of users from being able to access the fruits of everybody's labor here. Surely you can at least agree that is counterproductive?

sajanp, 2020-07-18:
> > @jaylittle just for sake of comparison, how much did you pay for your SLA with balena?
> 
> Not a single penny. But having this PPA down prevents a lot of users from being able to access the fruits of everybody's labor here. Surely you can at least agree that is counterproductive?

I'll agree it's counterproductive, buy only in the same way as your original comment about dropping and using an alternative over this.

jaylittle, 2020-07-18:
> I'll agree it's counterproductive, buy only in the same way as your original comment about dropping and using an alternative over this.

Is it?  The PPA is inaccessible.  As a Pop! OS Linux user, that puts me in an awkward position.  Sure it was already installed, but cannot be updated.  Packages that can't receive security updates do my security posture no favors.  YMMV of course.

fcastilloec, 2020-07-18:
Can somebody please close this issue, or at least lock the conversation, since nothing else added here helps with the issue itself. We could just mark this as duplicate of #3246

steelcowboy, 2020-07-18:
@fcastilloec This issue came first and has more comments than the issue you referenced

lurch, 2020-07-19:
I guess if anybody is super-concerned about their "security posture", they always have the option of manually downloading and installing the packages from https://github.com/balena-io/etcher/releases , until the problems with the bintary repository get sorted out? :man_shrugging:

PauloCreto-Linux, 2020-07-19:
I'm Brazilian user

BigLinux 20.04 ubuntu based

It has the same error that appeared a few days ago and the 3078 reports were closed but the problem persists

```
creto@creto:~$ sudo apt update
[sudo] senha para creto: 
Atingido:1 http://linux.teamviewer.com/deb stable InRelease
Atingido:2 http://dl.google.com/linux/chrome/deb stable InRelease                         
Atingido:3 http://deb.xanmod.org releases InRelease                                       
Atingido:4 https://deb.opera.com/opera-stable stable InRelease                            
Ign:5 http://linux.dropbox.com/ubuntu cosmic InRelease                                    
Atingido:7 https://download.virtualbox.org/virtualbox/debian eoan InRelease               
Obter:8 http://linux.dropbox.com/ubuntu cosmic Release [6.600 B]                          
Atingido:9 http://archive.canonical.com/ubuntu focal InRelease                            
Atingido:10 http://ppa.launchpad.net/system76/pop/ubuntu focal InRelease                  
Ign:11 http://biglinux.lua.im/mirror/biglinux/packages focal InRelease                    
Atingido:12 http://biglinux.lua.im/mirror/biglinux/packages focal Release                 
Atingido:13 http://ppa.launchpad.net/lutris-team/lutris/ubuntu focal InRelease            
Atingido:14 http://ubuntu.lua.im/ubuntu focal InRelease                                   
Atingido:15 http://ubuntu.lua.im/ubuntu focal-updates InRelease              
Atingido:16 http://ubuntu.lua.im/ubuntu focal-security InRelease             
Ign:6 https://dl.bintray.com/etcher/debian stable InRelease                  
Err:17 https://dl.bintray.com/etcher/debian stable Release
  403  Forbidden [IP: 52.36.128.157 443]
Lendo listas de pacotes... Pronto                                                         
E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
creto@creto:~$ 

```

PauloCreto-Linux, 2020-07-19:
> I guess if anybody is super-concerned about their "security posture", they always have the option of manually downloading and installing the packages from https://github.com/balena-io/etcher/releases , until the problems with the bintary repository get sorted out?

The case is not to download the .DEB, because the balenaEtcher works very well here on my BigLinux and only the update through the official repository, that's all

cadesalaberry, 2020-07-19:
I am having the same issue on Fedora:

```
➜ sudo dnf update --refresh
etcher-rpm                                                                                                                                                                                                     51  B/s |  10  B     00:00    
Errors during downloading metadata for repository 'etcher-rpm':
  - Status code: 403 for https://dl.bintray.com/etcher/redhat/repodata/repomd.xml (IP: 52.29.63.201)
Error: Failed to download metadata for repo 'etcher-rpm': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
```

thundron, 2020-07-20:
We're working on the issue, thank you all for the reports. We'll write here and close when we have updates, it's not a difficult one but might require some time as it involves some communication

thundron, 2020-08-03:
This should be solved now, closing and unlocking in case someone sees the error again.

closed 2020-08-03 by thundron

Topoignaz, 2021-05-14:
Today I got the same error during an update of my Debian Buster

Reading package lists... Done                                                                        
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 35.156.121.1 443]
E: The repository 'https://deb.etcher.io stable InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

lurch, 2021-05-14:
@Topoignaz see #3501

pakair, 2021-05-16:
Not resolved.  
                                                                                                                             
Reading package lists... Done
E: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  403  Forbidden [IP: 52.35.132.179 443]
E: The repository 'https://deb.etcher.io stable InRelease' is no longer signed.

In the balena forum there is a curl line to bring in the new PPA, it fails.
... distro=windowsfx  version=10  codename=ulyana  arch=x86_64

lurch, 2021-05-16:
> In the balena forum there is a curl line to bring in the new PPA, it fails.
... distro=windowsfx version=10 codename=ulyana arch=x86_64

Perhaps you need to get in touch with Cloudsmith to ask them to add support for "distro=windowsfx" ? :shrug: