balena-io/etcher #3245: Etcher should have FAQ entries for Windows displayed if a burn fails

saintaardvark, 2020-07-15:
A lot of questions come in on the forums that come down to trying to troubleshoot Windows issues.  It would be nice if Etcher had an FAQ for these kinds of problems ("try changing these Windows Defender settings", "try running as Admin"), and displayed them to the user if a write failed.  For bonus points, you could present a list, have the user check things off as they try them, and fall back to redirecting them to the forum.

lurch, 2020-07-16:
Seems there are also many Windows users who get confused by Windows not displaying the Linux-specific partitions in many of the disk images that Etcher is used for. (and/or only showing the small "boot" partition, which makes some Windows users complain that Etcher has reduced the size of their USB drive)
https://github.com/balena-io/etcher/issues/3091#issuecomment-594455533 :wink:

shawaj, 2020-12-13:
Perhaps could be added here? https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md