balena-io/etcher #3246: Balena Etcher Repository returns 403 Forbidden

Lamera, 2020-07-16:
- **Etcher version:** 1.5.100
- **Operating system and architecture:** Ubuntu 20.04

The Etcher repository is returning this error for a while.

`Ign:8 https://dl.bintray.com/etcher/debian stable InRelease`
`Fehl:12 https://dl.bintray.com/etcher/debian stable Release                                        
  403  Forbidden [IP: 3.120.60.234 443]`

**I tried the following:**

- Removed the application
- Removed the apt key
- Removed the source in /etc/apt/sources.d/
- Tried again to install with: [https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64)

BiNh0X, 2020-07-16:
Same problem in Linux Mint 19.3

schmidtw, 2020-07-17:
Same problem for Fedora 32.
All the files are forbidden here:
https://dl.bintray.com/etcher/redhat/

The directory structure appears & is navigable, but the content appears to have the wrong file permissions.

Luki72, 2020-07-17:
Same problem on Debian testing.
There appears to be a file permission error on the server.

steelcowboy, 2020-07-18:
@Lamera please close this issue, it is a duplicate of #3244

closed 2020-07-19 by Lamera