balena-io/etcher #3251: ubuntu/debian ppa failed to install

Abinash393, 2020-07-19:
Latest 
# Ubuntu 18.04
- **Image flashed:** 

Ign:8 https://dl.bintray.com/etcher/debian stable InRelease                                                                        
Err:17 https://dl.bintray.com/etcher/debian stable Release
  403  Forbidden [IP: 52.36.128.157 443]

lurch, 2020-07-19:
Duplicate of #3244

cyberjew, 2020-07-19:
I'm running into the same issue but I am running Ubuntu 20.04. This issue appears to be related to bintray.com and is not operating system version specific.

I used Firefox to browse to https://deb.etcher.io. There was an HTTP 302 redirect to https://dl.bintray.com/etcher/debian/. You can see a list of Debian packages for Etcher in that folder, however if you navigate to https://dl.bintray.com/etcher/debian/dists/stable/ and click on the 'Release' or 'Release.gpg' file, the contents say "Forbidden".

Based on the number of other issues of similar nature that have been reported in the past, if I were in Balena's shoes, I would find another hosting provider and kick bintray.com to the curb.

Abinash393, 2020-07-20:
#### As for now the below method works for me. Fixing the ppa is up to the maintainers
- go to the website and choose the latest version
```
https://github.com/balena-io/etcher/releases
```
- download the package for your OS and Architecture for me it's
```
balena-etcher-electron_1.5.101_amd64.deb
```
- open the terminal and go to Downloads and install
```bash
cd Downloads
sudo dpkg -i balena-etcher-electron_1.5.101_amd64.deb
```

closed 2020-07-20 by Abinash393