balena-io/etcher #3253: 403 Forbidden error on APT package repository (again)

tomreyn, 2020-07-20:
- **Operating system and architecture:** 
Ubuntu 18.04 LTS

As in #3078, accessing the Bintray apt repository fails again with a 403:
```
$ sudo apt update
 ...
Err:21 https://dl.bintray.com/etcher/debian stable Release                                                                                                    
  403  Forbidden [IP: 52.29.63.201 443]
 ...
Reading package lists... Done                                                           
E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

$ apt policy balena-etcher-electron 
balena-etcher-electron:
  Installed: 1.5.100
  Candidate: 1.5.100
  Version table:
 *** 1.5.100 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
        100 /var/lib/dpkg/status
     1.5.76 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.5.51 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.5.19 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
     1.4.9 500
        500 https://deb.etcher.io stable/etcher amd64 Packages
```

lurch, 2020-07-20:
Duplicate of #3244

tomreyn, 2020-07-20:
Thanks, lurch, and sorry about the dupe - closing.

closed 2020-07-20 by tomreyn