balena-io/etcher #3254: Problem with Kali Linux

Datalva, 2020-07-20:
Hi,
When I try to install in Kali Linux 2020, say

tecnico@kali:~$ sudo apt-get update
[sudo] password for tecnico: 
Obj:1 http://mirror.cedia.org.ec/kali kali-rolling InRelease
Ign:2 https://dl.bintray.com/etcher/debian stable InRelease
Err:3 https://dl.bintray.com/etcher/debian stable Release
  403  Forbidden [IP: 52.36.128.157 443]
Leyendo lista de paquetes... Hecho
E: El repositorio «https://deb.etcher.io stable Release» no tiene un fichero de Publicación.
N: No se puede actualizar de un repositorio como este de forma segura y por tanto está deshabilitado por omisión.
N: Vea la página de manual apt-secure(8) para los detalles sobre la creación de repositorios y la configuración de usuarios.

lurch, 2020-07-20:
Duplicate of #3244

thundron, 2020-07-21:
Please refer to the issue above. Closing

closed 2020-07-21 by thundron