balena-io/etcher #3259: apt error

Adam-vk2pwr, 2020-07-21:
- *github * 
- **linux mint 20** 
- **Image flashed:none** 
- **Do you see any meaningful error information in the DevTools?no ** 

Err:22 https://dl.bintray.com/etcher/debian stable Release       
  403  Forbidden [IP: 35.161.162.245 443]
Reading package lists... Done
E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

lurch, 2020-07-22:
Another duplicate of #3244

Adam-vk2pwr, 2020-07-22:
> Another duplicate of #3244

well is there a fix

thundron, 2020-07-22:
Not yet, but we're working on it. In the meantime you can download through the github releases page here https://github.com/balena-io/etcher/tags

closed 2020-07-22 by thundron