balena-io/etcher #3261: Installation Frustration

noobTrying2learn, 2020-07-23:
- **1.5.101** 
- **Linux Mint LMDE 4, x86_64 ** 
- **?? I followed scripts in the github repository** 
- **?? no** 

I have followed the instructions located [here](https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64), but they get me nowhere. I have searched for the error message following the apt-get update command to find no help. here is the output following the "sudo apt-get update" command

 E: The repository 'https://deb.etcher.io stable Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

again, I have searched diligently (over an hour) for a solution only to be going in circles. 

I like balena-etcher on my mac and would love to run it on my linux box, but it seems it's a lost cause. I'm not a complete idiot when it comes to computers, but I am far from an expert. Please, if someone can help I would be very appreciative.

lurch, 2020-07-23:
Duplicate of #3244
I believe in the meantime you can download the .deb file directly from https://github.com/balena-io/etcher/releases

thundron, 2020-07-23:
@noobTrying2learn Sorry for the hassle, it's an issue on our side (to be precise, our package host who we're in contact with to solve the issue) which we're tracking on #3244 - in the meantime you can download the debian from the releases page here on Github https://github.com/balena-io/etcher/tags

closed 2020-07-23 by thundron