balena-io/etcher #3262: Etcher takes few minutes to start and show GUI on macOS

jiribrejcha, 2020-07-23:
After starting Etcher app, it takes few minutes for its window to actually appear on the screen. After it appears, it then works fine.

- **Etcher version:** 
Version 1.5.101 (1.5.101)

- **Operating system and architecture:** 
macOS Catalina 10.15.6 (19G73)

- **Image flashed:** 
Not even got to the GUI yet.

- **Do you see any meaningful error information in the DevTools?** 
No

lurch, 2020-07-23:
Is this the same as #2740 ?

jiribrejcha, 2020-07-23:
No, this is different. I see this issue only at startup of the app. After it starts (takes a few minutes), it then works fine and GUI works well and refreshes fast.

jiribrejcha, 2020-07-29:
It suddenly started working after a few days.

closed 2020-07-29 by jiribrejcha