balena-io/etcher #3266: Debian repository dead

Tarnatos, 2020-07-26:
Hey,

are you planing to fix the debain repository?

https://dl.bintray.com/etcher/debian/dists/stable/Release

Results in "Forbidden!"

Thx in advance
Tarnatos

martin-schlossarek, 2020-07-26:
Same for The Red Hat Repo: https://dl.bintray.com/etcher/redhat/repodata/repomd.xml

lurch, 2020-07-26:
Duplicate of #3244

djvar94, 2020-07-26:
Yet again, the bloody repository is down. My God, how many times more do you have to keep breaking the repository? Honestly, so annoying, you are the only software I know that REGULARLY keeps breaking the repos.
Fix this once and for all please.

thundron, 2020-07-27:
First of all, I think we can all agree on keeping a civilised tone for the discussion. We're all adults and there should be no need to clarify that, but better clear that out of the way.
Second, it seems that we're being limited on the package host side, which is an external service we rely on.
Third, we're in contact with said package host to solve the issue but communications are somewhat slow, so patience is the key.
Fourth and last, I'm assuming you already have etcher installed so you should be set until we solve the issue - if not, you can temporarily download directly from our github releases https://github.com/balena-io/etcher/tags which are the same files we upload on the deb/rpm repositories.

Thanks for the patience and understanding, closing as duplicate.

closed 2020-07-27 by thundron

martin-schlossarek, 2020-07-27:
> First of all, I think we can all agree on keeping a civilised tone for the discussion. We're all adults and there should be no need to clarify that, but better clear that out of the way.
> Second, it [...]

It is still really annoying because this will throw errors everytime you run yum or apt.

djvar94, 2020-07-27:
> It is still really annoying because this will throw errors everytime you run yum or apt.

EXACTLY MY POINT. PERFECTLY SPOT ON! And what's worse for these guys is that it's not the first time this happens. Which is so so much more annoying. And don't come and tell me to be calm in my conversation when you've been running this total nonsense for LITTERALLY MONTHS! You have problems with your third party provider? Your communications are slow? Then bloody change 3rd party provider for goodness sake. Like everyone else does. So many one person developers and software engineers use repositories like on Launchpad and they NEVER CAUSE MY SYSTEM TO SPIT OUT ERRORS ON REPOS LIKE YOU DO! As I said before you are the only one that give this s#!t. But you're working on it. Yeah, I've been waiting at least two months now already and still the repo is broken. And still it annoys me every time I check for updates in apt or system update manager. If you keep up with this pace the current human generation will be dead by the time you fix this. As I said earlier, you're far better off changing 3rd party repository provider altogether and not endure this complete embarrassment.

thundron, 2020-07-27:
> It is still really annoying because this will throw errors every time you run yum or apt.

I can understand the frustration and it comes by both sides, trust me, it's not that we're happy when there's something going wrong with our ecosystem, but there's only so much we can do in certain situations.