balena-io/etcher #3267: Does not work on Linux Mint 20

cayandris, 2020-07-29:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

thundron, 2020-07-30:
Could you elaborate more on the issue and fill in the information of the template?

StephanJanosch, 2020-08-19:
Same here. Tried to use 
* Etcher version 1.5.102 
* on Mac OS 10.14.6
* Image, Mint 20: https://linuxmint.com/edition.php?id=282

Solution. Downgraded Etcher to 1.5.56 (picked randomly) and image was flashed flawless.

StephanJanosch, 2020-08-19:
After using Etcher 1.5.56 once now 1.5.102 works fine as well. Not sure if that helps. I could remove some prefs and retest if I would know where to look for the preferences.

Jim-Walk, 2020-11-16:
This issue is also affecting me, linux mint 20, Mac OS 10.15.4. Etcher 1.5.110 fails, as does 1.5.109. Using a dongle with on a 2019 mac and a micro sd card

thundron, 2020-11-17:
This issue is for linux mint 20, not for macOS. Closing for lack of information from op

closed 2020-11-17 by thundron

lurch, 2020-11-17:
@Jim-Walk To clarify, the original issue was about running Etcher on Linux Mint itself, **not** using Etcher to flash a Linux Mint image.
If you have a problem running Etcher on macOS, you should create a new issue, including as much info as you can :)