balena-io/etcher #3269: Unable to package using electron-Build

btwise, 2020-07-31:
- **Etcher version:1.5.102
- **Operating system and architecture:WIN10 

The latest 1.5.102 was constructed normally with ` **make electrone-develop**` under WINODWS10 and operated normally with `npm start`, but packaging could not be completed withe **`electrone-build`**. How can we solve the following errors?

[image]

shawaj, 2020-12-15:
@btwise is there a reason you are compiling yourself rather than just downloading the installer?

And are you using electron-build? Was the extra e at the end of electron in the above comment?

lurch, 2020-12-16:
Looking at the screenshot, I wonder why it's trying to run `bash` on Windows? :shrug:

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt