balena-io/etcher #3272: Etcher ZIP file downloads via Firefox almost every 30 minutes

engineer22, 2020-08-03:
I just noticed that a release zip file downloads or tries to download via firefox, almost every 30 minutes. Sometimes it appears on the downloads menu inside firefox without actually being downloaded, and sometimes it will state a 'download failed' and as soon as I click retry, it will show up on the downloads folder without taking time to download.

The file shows up even when I don't run APT updates, and if I run an update it won't make it appear if around 30 minutes have passed.

- system: debian 10.5 (stable)
- install method: by adding key and downloading via APT
- automatic updates: off

thundron, 2020-08-03:
@engineer22 We haven't got any retry logic similar to that on our website and the application updates are handled in the application itself + not every 30mins, could it be a tab that you left open that has the download link? What's the name of the package that is being downloaded?

engineer22, 2020-08-03:
@thundron its definitely not a tab that forgoten, I cleared everything and even the timestamp changes.

Aftet couple of tests, I came to conclusion that this happens only when I enable 'automatic updates' within the UI settings.

I will re-enable the setting and update you eith the filename when it generates again.

lurch, 2020-08-03:
I wonder if it might be some kind of weird web proxy interaction? :shrug: (but even so, that would still be _very_ strange)
How is your computer connected to the internet?

engineer22, 2020-08-05:
@lurch nothing special. The only thing modified is the DNS (Cloudflare)!

@thundron after uninstalling etcher, removing the key and any trace, I reinstalled it and until now the problem doesn't exist.

Since there are no other issues related to this, I will go ahead and close the issue. Thank you, both!

closed 2020-08-05 by engineer22