balena-io/etcher #3275: Flashing USB of 64GB with 25GB ISO file taking almost 4hours via etcher CLI.

ndunaboi, 2020-08-08:
- **Etcher version:** 
1.2.0 and 1.1.2  Etcher-cli-1.2.0-linux-x64

- **Operating system and architecture:** 
$ uname -a
Linux  4.15.0-106-generic #107~16.04.1-Ubuntu SMP  x86_64 x86_64 x86_64 GNU/Linux

- **Image flashed:** 

- **Do you see any meaningful error information in the DevTools?** 
No i am using CLI. i dont see any error messages. but just the process is way too slow. where as Etcher UI on macbook is taking 30min for same USB.

lurch, 2020-08-09:
Not directly relevant to Etcher-cli, but see also #2851

P.S. Does it _actually_ take 4 hours to write the file, or is it just that the initial time-estimate is way off?

thundron, 2020-08-10:
The etcher-cli has been long discontinued, please use the Etcher application or the etcher-sdk examples.

closed 2020-08-10 by thundron