balena-io/etcher #3276: Etcher killed my 2 usb sticks in a single day

3axap4eHko, 2020-08-11:
- **Etcher version: 1.5.102** 
- **Operating system and architecture: Windows 10 x64 2004** 
- **Image flashed: ubuntu-20.04.1-desktop-amd64.iso** 
- **Do you see any meaningful error information in the DevTools? NO** 

I started writing ubuntu image on the first drive and it was throwing me an error
[image]
Now it's dead. Once I tried to use the second drive it did the same.
So I have 2 broken drives.

thundron, 2020-08-12:
This can be due to several reasons (which are not related to Etcher), for example Windows locking your drive so it's write-protected, along with other.
I'd suggest looking for the error code you see in the Event Log (`5@0101000F`) on the Microsoft forums to further investigate.
On the subject of "etcher killing USBs" please see https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/

closed 2020-08-12 by thundron

lurch, 2020-08-12:
@thundron Nice article - probably worth adding a link to it from the Etcher README? :slightly_smiling_face:

Uranite, 2021-07-23:
https://youtu.be/UjFEKpeVc1s