balena-io/etcher #3278: Etcher not showing up on macOS Catalina

t0mm4rx, 2020-08-16:
- **Etcher version:** 1.5.102
- **Operating system and architecture:** macOS Catalina 10.15.3
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** DevTools are blank

Hello,
I'm trying to use Etcher on macOS, but when I launch the app nothing happen.
I've checked devtools, it's blank.
I've tried running the app as sudo, nothing happen, the terminal is blank too.
When I click on "Show all windows", it says there is none.
Other Electron apps are working well on my computer.

shawaj, 2020-12-08:
Could you try with the latest version v1.5.112 @t0mm4rx ?

https://www.balena.io/etcher/

t0mm4rx, 2020-12-16:
It's working !

closed 2020-12-16 by t0mm4rx

shawaj, 2020-12-16:
@t0mm4rx awesome thanks for the update!