balena-io/etcher #3280: Cannot select drive.

DawiDMAJAJA3, 2020-08-20:
- **Etcher version:** v1.5.103
- **Operating system and architecture:** Win10 (64)
- **Image flashed:** I did not flash image. 
- **Do you see any meaningful error information in the DevTools?** No

So basically, when I want to boot up with Kaspersky Rescue Tool, I cannot select drive (Screenshot). Help would be appreciated.

[image]

thundron, 2020-08-20:
You're trying to flash your system drive, which is dangerous unless *you know what you're doing*, and in this case it's the source of your image so you can't use it as a target.

closed 2020-08-20 by thundron

sajolida, 2021-12-22:
Let's see if we can fix the root cause of bad feedback on #3662.