balena-io/etcher #3285: Feature request: total copies countdown

sboutros, 2020-08-26:
It would be awesome to have a feature to specify how many copies of an image I need, say 12.
Then as I hit start and walk away and periodically check back, pull cards out, put new cards in, and invariably lose track of which card I am on.
Keep a count-down timer till I have 12 successfully validated SD cards…

Another poster suggested tracking specific cards as well to prevent accidental re-imaging:
Just a simple locally stored history of the UUID (if there is such a thing) of the drives flashed and which image you flashed to each of them. Is there a UUID that is hardware based for sdcards, emmc and usb drives?

You could then count how many flashes you have completed in this session and warn you if you about to re-flash the same drive twice (in the same etcher instance session). (Done this so many times) Normally I try and flash with 4 versions of etcher running at once because I found that more stable than multi targeted flashes.

lurch, 2020-08-27:
> Then as I hit start and walk away and periodically check back, pull cards out, put new cards in, and invariably lose track of which card I am on.

You could just keep "unflashed cards" and "cards that have been flashed" in two separate (labelled) pots? :wink: 

> Is there a UUID that is hardware based for sdcards

SD cards have a unique [CID register](https://www.cameramemoryspeed.com/sd-memory-card-faq/reading-sd-card-cid-serial-psn-internal-numbers/), but that's _only_ readable on SD-card-readers that are connected via PCI (i.e. that appear as `/dev/mmcblk0` under Linux, not SD-card-readers that are connected via USB (i.e. that appear as `/dev/sdX` under Linux).
(some laptops have their internal card-readers connected via USB, some laptops have their internal card-readers connected via USB)

> It would be awesome to have a feature to specify how many copies of an image I need, say 12.

How is [EtcherPro](https://www.balena.io/etcher/pro/) coming along @thundron / @zvin ? :slightly_smiling_face:

thundron, 2020-08-27:
@lurch Soon™ 

> tracking specific cards as well to prevent accidental re-imaging

> Keep a count-down timer till I have 12 successfully validated SD cards

@sboutros you can actually connect 12 targets and flash them all at once, this was implemented a long time ago: it already both saves the time of re-connecting the targets and the hassle of knowing which card was already flashed (given you don't put them back with the un-flashed ones of course).

Having a step-by-step flash procedure would be redundant since multi-writing is already implemented and keeping track of the images flashed wouldn't be as accurate if, for example, you want to flash them again (again, given that multi-writing already exists).

closed 2020-08-27 by thundron