balena-io/etcher #3287: Can we start decompressing image before elevating permission?

chrisys, 2020-08-27:
Right now when you click flash, Etcher asks to elevate permissions in order to write to the flash drive, then afterwards starts decompressing ready to flash. Could we start the decompression process whilst it is waiting for the permission to write to the drive? Meaning that when/if permission is obtained, the image is ready to flash immediately, thereby optimising the whole flashing process a little more.

thundron, 2020-08-27:
It's possible, but I don't think it's worth doing - we use a temporary file for decompression that we generate every time we start the flashing process and we get permission right before running the child-process script so we would need to handle that logic in both the application *and* etcher-sdk.
@zvin What do you think?

lurch, 2020-08-27:
I think the bottleneck is usually in the SD card write-speed rather than the image-decompression speed, so it doesn't seem likely that this approach would offer much benefit? (IIRC Etcher decompresses and writes in parallel, right @thundron ?)

thundron, 2020-08-27:
@lurch We switched to decompressing first (though it can be toggled on the etcher-sdk level)

lurch, 2020-08-27:
Ahhh. Do you mind if I ask what the reason for that was? (just out of curiosity)

chrisys, 2020-08-31:
Thanks @thundron 

It is around 5 seconds to decompress in the case of a balenaOS image, but when the total flashing time is ~25 seconds it could be a 20% improvement assuming the user takes longer than 5 seconds to give permission.