balena-io/etcher #3289: Etcher should warn that the disk will be formatted or erased

pdcastro, 2020-08-28:
Many users (especially those in the IoT community) understand that "Flashing" a disk means reformatting / erasing it, but not all users do (see attached support thread). Etcher uses only the term "flashing" on the GUI, whereas more commonly, the terms "formatting" and "erasing" are used (especially respectively on Windows and macOS).

Perhaps a confirmation pop-up could be raised when the button "Flash!" is clicked, and this pop-up could read something along the lines of:

> 
> **Flashing the selected drive will reformat / erase it, and all existing data on the drive will be lost.**
>
> [ Confirm ]     [ Cancel ]
> 
> [✓] Don't show this prompt again for 30 days
> 

- **Etcher version:** 1.5.106
- **Operating system and architecture:** any (macOS 10.14.6)
- **Image flashed:** any
- **Do you see any meaningful error information in the DevTools?** N/A

lurch, 2020-08-29:
See also #3049

FloMiau, 2020-09-16:
Nobody likes warnings. Maybe call the button "Erase and Flash" or something like that.

dtischler, 2020-10-16:
Adding an additional data point here, as a user made this same request in the comments on one of our blog posts, http://disq.us/p/2bzxux1

jharris1993, 2020-10-17:
> Nobody likes warnings. Maybe call the button "Erase and Flash" or something like that.

Au contrare!

What people *don't* like is to see valuable data go "poof!" because someone doesn't know that "flash" means "nuke-and-destroy".

Warnings are annoying, but - like sudo - they exist for a reason.

Additional thought:
Placing a "hide the warning" selector within the same dialog is a recipe for disaster. All it takes is a slight tremor on a touch-screen, an errant palm rub on a touch-pad, or a three year old bumping the hand holding the mouse to send an important and necessary warning into nowhere-land.

A better choice is to make it a configuration setting such as:
[ X ] Always warn
[    ] Warn only once during a session
[    ] Never warn

This let's the advanced user punt on the warnings, a somewhat advanced user opt for a single warning, and the non-technical user get all warnings.  "Always warn" should be the default.

jharris1993, 2026-02-11:
> I think that it's quite given that it will erase the USB drive or SSD card that you're using for the Etcher.

Famous last words. . . .

Why do airplane pilots have checklists?  Because what might be "quite given" for you may not be so obvious to someone else.

AFAIK, it's a generally accepted best-practice to warn the user/operator when they're about to do something irrecoverably destructive.

As an example:

The "YUMI" bootable flash-drive creator from PendriveLinux, (https://pendrivelinux.com/yumi-multiboot-usb-creator/#yumi-exfat-download), makes it in-your-face obvious that what you're going to do will nuke the content of the attached flash drive into never-never land.

Window's Media Creation Tool also explicitly warns the user prior to nuking the target device.

This is so common that it is essentially a _de facto_ industry standard.  So much so that I am actually surprised that we're even having this discussion.