balena-io/etcher #3290: Etcher stuck at "Decompresing"

crysisrules, 2020-08-28:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

I trying to flash image to SD card but it stuck at " Decompressing...100%" What is the problem can someone help me to solve this? I had tried different  laptop and PC and also tried different SD card but no luck. Please help. Thanks.

[image]

[image]

CarnageMarkus, 2020-09-10:
I have the same problem

lurch, 2020-12-14:
I dunno what might be causing this, but perhaps the "temporary workaround" I suggested here might help? https://github.com/balena-io/etcher/issues/3164#issuecomment-744150863
(i.e. decompress the image outside of Etcher)

hater01, 2022-07-16:
i have a raspberry pi and i had the same problem when i tried to install raspbian. but then i extracted all the files from the rar file and then i put the new iso file on the sd and it worked