balena-io/etcher #3291: Consider releasing less often?

giorgiga, 2020-08-30:
For a while I had the impression that every time I updated homebrew a new version of etcher would be (paifully slowly) downloaded... now that I took a look at the release and commit histories in this repo I can see why.

I already uninstalled etcher and will proceed to reinstall next time I needed and uninstall again shortly thereafter and so, in a sense, I already worked around this issue.

However I am sure I am not the only person inconvenienced by the continuous releases, so I would still like to suggest you devs if you think implementing some kind of release cycle would be a good idea (even weekly releases would be better than this).

lurch, 2020-08-30:
See also #3175

thundron, 2020-09-08:
We're in the process of changing the update system, we'll have a PR in the future but right now our focus is not there

closed 2020-09-08 by thundron