balena-io/etcher #3292: can not to build on MS-Windows

btwise, 2020-08-31:
- **Etcher version:1.15.102+
- **Operating system and architecture:win10

Tiffutil is a macOS program. When packaged on Windows using make electron-build, the error message that tiffutil cannot be found appears!

lurch, 2020-09-01:
Looking at the `Makefile` it [seems like](https://github.com/balena-io/etcher/blob/master/Makefile#L103) `electron-build` is unconditionally depending on `assets/dmg/background.tiff` (and `tiffutil` is needed to make `background.tiff`).
However I guess `electron-build` should probably only depend on `assets/dmg/background.tiff` when the `$(PLATFORM)` is `darwin` ? :man_shrugging: 

Hmmm, although `background.tiff` _does_ already exist in https://github.com/balena-io/etcher/tree/master/assets/dmg so in theory there should be no need to re-make it?

thundron, 2020-11-24:
I built fine on Windows following the contributing file and while it misses some things, I had no issues with cross-platform packages.
What error do you see when building?

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)