balena-io/etcher #3293: AppImage 1.5.106-x64 - Sandbox not configured correctly

as365n4, 2020-09-04:
- **Etcher version:** balenaEtcher-1.5.106-x64.AppImage
- **Operating system and architecture:** Debian 10.5 stable
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** n/a

Since Etcher can't be updated through apt, I had purged Etcher from my System and downloaded the AppImage from balena.io
AppImage is marked as executing file, but any attempt in launching the app fails with a big great nothing as the app won't start.
Launching the AppImage from the termianl results in the following error message:`[7718:0904/112132.674750:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenaKfRiHD/chrome-sandbox is owned by root and has mode 4755.
/tmp/.mount_balenaKfRiHD/balena-etcher-electron: line 10:  7718 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"`

lurch, 2020-09-04:
Seems to be a few other people with this issue too - see https://github.com/balena-io/etcher/search?q=sandbox&type=Issues

thundron, 2021-01-15:
Actually, this issue https://github.com/balena-io/etcher/issues/3129 has some more info in it and is probably better for that reason to use as tracking. Moving there

closed 2021-01-15 by thundron