balena-io/etcher #3296: Etcher not starting due to virus/ransomware protection tools

sradevski, 2020-09-11:
A user had issue with Etcher's GUI not starting up due to "Acronis Active Protection", which is a ransomware protection tool, blocking it. There might not be anything we can do, and it might generalize to all Electron apps, so opening in case we can somehow mitigate the problem.

thundron, 2020-09-14:
The user needs to add an exception for Etcher in his protection tools, this is a known issue which unfortunately has to be handled user-side and happens with other protection tools as well.
"Back in the day" when the developer certificate was new even Windows was skeptical about Etcher but that eventually went away with Etcher being used more and more (which IIRC is how Windows evaluates new developer certificates/apps outside of the Microsoft Store)

closed 2020-09-14 by thundron

lurch, 2020-09-15:
Yep, see https://github.com/balena-io/etcher/search?q=acronis&type=issues and https://github.com/balena-io/etcher/search?q=whitelist&type=issues and https://github.com/balena-io/etcher/search?q=defender&type=issues for similar issues.