balena-io/etcher #3299: Portable Version won't etch: Something wen't wrong

Josephur, 2020-09-12:
- **Etcher version:** Portable 1.5.108
- **Operating system and architecture:** Windows 10 (x64)  2004 19041.508
- **Image flashed:** https://octopi.octoprint.org/latest
- **Do you see any meaningful error information in the DevTools?** 

gui.js:48918 {"stack":"Error: Command failed: cmd /c \"C:\\Users\\JOSEPH~1.STA\\AppData\\Local\\Temp\\balena-etcher-electron--17576-EsMcNUUs525B-.cmd\"\r\n\n    at file:///C%3A/Users/JOSEPH~1.STA/AppData/Local/Temp/1hJiTKxQC5OF9PkZrCupnveh2JT/resources/app/generated/gui.js:29102:33\n    at FSReqCallback.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:61:3)","message":"Command failed: cmd /c \"C:\\Users\\JOSEPH~1.STA\\AppData\\Local\\Temp\\balena-etcher-electron--17576-EsMcNUUs525B-.cmd\"\r\n","code":-1073741819,"image":"2019-09-26-octopi-buster-lite-0.17.0.img"}

Note that if I install etcher using the installation and not the portable version, the issue goes away.  Must be a problem with how it extracts to a temp folder or something.  I tried running as admin, using diskpart to clear the flash drive first.. did not matter.  Etching the same image using the setup version of etcher works flawlessly.

goseeped, 2020-09-12:
I'm in Windows 10 x64 the same issue,  Version = 1.5.108, Type = portable

Here are my logs if that helps.
```
Sat Sep 12 2020 13:49:53 GMT-0700 (Pacific Daylight Time) Application start ({"packageType":"portable","version":"1.5.108","sample":0.1,"applicationSessionUuid":"f777b809-04bb-4648-9277-fb8c25b89d47","flashingWorkflowUuid":"3f4ed8de-073e-4491-8a12-8766b6c10cfd"})
gui.js:48918 Sat Sep 12 2020 13:50:12 GMT-0700 (Pacific Daylight Time) Open image selector ({"sample":0.1,"applicationSessionUuid":"f777b809-04bb-4648-9277-fb8c25b89d47","flashingWorkflowUuid":"3f4ed8de-073e-4491-8a12-8766b6c10cfd"})
gui.js:48918 Sat Sep 12 2020 13:50:16 GMT-0700 (Pacific Daylight Time) Select image ({"image":{"size":2166359040,"name":"PrusaPrint.img","hasMBR":true,"partitions":[{"offset":4194304,"size":43832832,"type":12,"index":1},{"offset":50331648,"size":2116027392,"type":131,"index":2}],"extension":"img","path":"S:\\3DPrinter\\OctoPrint\\PrusaPrintIMG\\PrusaPrint.img","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"f777b809-04bb-4648-9277-fb8c25b89d47","flashingWorkflowUuid":"3f4ed8de-073e-4491-8a12-8766b6c10cfd"})
gui.js:48918 Sat Sep 12 2020 13:50:23 GMT-0700 (Pacific Daylight Time) Toggle drive ({"drive":{"size":32010928128,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"E:\\"}],"description":"Lexar uSD UHS2 RDR USB Device","isUAS":false},"previouslySelected":false,"sample":0.1,"applicationSessionUuid":"f777b809-04bb-4648-9277-fb8c25b89d47","flashingWorkflowUuid":"3f4ed8de-073e-4491-8a12-8766b6c10cfd"})
gui.js:48918 Sat Sep 12 2020 13:50:37 GMT-0700 (Pacific Daylight Time) Flash ({"image":{"size":2166359040,"name":"PrusaPrint.img","hasMBR":true,"partitions":[{"offset":4194304,"size":43832832,"type":12,"index":1},{"offset":50331648,"size":2116027392,"type":131,"index":2}],"extension":"img","path":"S:\\3DPrinter\\OctoPrint\\PrusaPrintIMG\\PrusaPrint.img"},"drives":[{"size":32010928128,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"E:\\"}],"description":"Lexar uSD UHS2 RDR USB Device","isUAS":false}],"driveCount":1,"uuid":"0c5ea182-64f3-4f5b-9973-c5863892ac1b","status":"started","flashInstanceUuid":"0c5ea182-64f3-4f5b-9973-c5863892ac1b","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1,"applicationSessionUuid":"f777b809-04bb-4648-9277-fb8c25b89d47","flashingWorkflowUuid":"3f4ed8de-073e-4491-8a12-8766b6c10cfd"})
gui.js:48918 Elevating command: C:\Users\goseeped\AppData\Local\Temp\1hJiTKxQC5OF9PkZrCupnveh2JT\balenaEtcher.exe C:\Users\goseeped\AppData\Local\Temp\1hJiTKxQC5OF9PkZrCupnveh2JT\resources\app\generated\child-writer.js
gui.js:48918 Successfully connected to IPC server: etcher-server-12980, socket root C:\Users\goseeped\AppData\Local\Temp\
gui.js:48918 Image: S:\3DPrinter\OctoPrint\PrusaPrintIMG\PrusaPrint.img
gui.js:48918 Devices: \\.\PhysicalDrive2
gui.js:48918 Umount on success: true
gui.js:48918 Validate on success: true
gui.js:48918 Auto blockmapping: true
gui.js:48918 Decompress first: true
gui.js:48918 Terminating IPC server
gui.js:48918 Flash results 
{}
gui.js:48918 Sat Sep 12 2020 13:50:37 GMT-0700 (Pacific Daylight Time) Write failed ({"image":{"size":2166359040,"name":"PrusaPrint.img","hasMBR":true,"partitions":[{"offset":4194304,"size":43832832,"type":12,"index":1},{"offset":50331648,"size":2116027392,"type":131,"index":2}],"extension":"img","path":"S:\\3DPrinter\\OctoPrint\\PrusaPrintIMG\\PrusaPrint.img"},"drives":[{"size":32010928128,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive2","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive2","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"E:\\"}],"description":"Lexar uSD UHS2 RDR USB Device","isUAS":false}],"driveCount":1,"uuid":"0c5ea182-64f3-4f5b-9973-c5863892ac1b","status":"failed","flashInstanceUuid":"0c5ea182-64f3-4f5b-9973-c5863892ac1b","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{"killed":false,"code":3221225477,"signal":null,"cmd":"C:\\Users\\goseeped\\AppData\\Local\\Temp\\1hJiTKxQC5OF9PkZrCupnveh2JT\\balenaEtcher.exe child-writer.js","stdout":"","stderr":""},"sample":0.1,"applicationSessionUuid":"f777b809-04bb-4648-9277-fb8c25b89d47","flashingWorkflowUuid":"3f4ed8de-073e-4491-8a12-8766b6c10cfd"})
gui.js:48918 {"stack":"Error: Command failed: C:\\Users\\goseeped\\AppData\\Local\\Temp\\1hJiTKxQC5OF9PkZrCupnveh2JT\\balenaEtcher.exe C:\\Users\\goseeped\\AppData\\Local\\Temp\\1hJiTKxQC5OF9PkZrCupnveh2JT\\resources\\app\\generated\\child-writer.js\n\n    at ChildProcess.exithandler (child_process.js:304:12)\n    at ChildProcess.emit (events.js:223:5)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","message":"Command failed: C:\\Users\\goseeped\\AppData\\Local\\Temp\\1hJiTKxQC5OF9PkZrCupnveh2JT\\balenaEtcher.exe C:\\Users\\goseeped\\AppData\\Local\\Temp\\1hJiTKxQC5OF9PkZrCupnveh2JT\\resources\\app\\generated\\child-writer.js\n","killed":false,"code":3221225477,"signal":null,"cmd":"child-writer.js","stdout":"","stderr":"","image":"PrusaPrint.img"}
```

thundron, 2020-09-14:
@Josephur @goseeped I know it sounds silly, but can you try version `v1.5.106` (**portable**) and report back the results?

Eenie75, 2020-09-14:
Hi,

I was having the exact same problem, and I tried the v1.5.106 Portable and I was able to flash the image successfully.

closed 2020-09-14 by bulldozer-balena[bot]

Josephur, 2020-09-15:
Just as a note, my username did not contain an ampersand, but it did contain a period which is common in Domain networks.  Like Firstname.Lastname