balena-io/etcher #3300: Native support for Apple silicon (without using Rosetta)

sradevski, 2020-09-14:
This is a placeholder issue for all the things that we might need to change to support Apple silicon natively once it's released.

thundron, 2020-09-14:
Electron devs are already on track with it https://github.com/electron/electron/issues/24319 so hopefully we won't need to change too many things, I think we haven't got many dependencies other than maybe `drivelist` which *could* need an upgrade, *if* some native calls change.
We'll see how it evolves, keeping this issue open for tracking purposes

thebitstick, 2020-11-23:
Apple Silicon Macs released a week ago, and Electron updated with support as well.

fluxtransistor, 2021-02-02:
It's been a little while, any word on the progress here?

thundron, 2021-02-02:
@fluxtransistor there are some bugs on electron 11 that prevent us upgrading, so it will need some more time until those issues are resolved (see e.g. https://github.com/electron/electron/issues/25425 )

fluxtransistor, 2021-02-02:
Alright, thanks!

thedocbwarren, 2021-02-04:
Seems like they have a workaround.  Is this something that can be implemented or part of an alpha to test this out?  My concern is running through rosetta that has device access.

lurch, 2021-03-29:
Apparently this is being added in #3432

closed 2021-05-26 by thundron

fishcharlie, 2021-11-08:
@thundron Why was this closed? It was **not** fixed in #3432. As of v1.6.0 the build is still Intel only. Can we reopen it to draw attention to the fact that this is not resolved?

thundron, 2021-11-08:
There was likely a regression which we are trying to figure out in the latest build so the issue is definitely not the same, @fishcharlie could you please check v1.5.122 for that matter? We also have reports of 1.6.0 still working with rosetta without emulation, despite not being tagged as so in the file information and I'm trying to gather as much information as possible

fishcharlie, 2021-11-08:
@thundron Still Intel only.

[image]

lurch, 2021-11-12:
Links to #3593

gandalfwfh, 2023-01-23:
2023 still rosetta rocks, users with no rosetta can use only stock apple software

trigger2k20, 2023-07-25:
July 2023 still rosetta, why ?

shareef777, 2023-12-12:
The latest Apple Silicon supported version is 1.5.121. Support seems to have been dropped after that.

trigger2k20, 2023-12-13:
But why ? this is not very logical.

Mit freundlichen Grüßen
Sascha Afkhampour

***@***.***