balena-io/etcher #3301: Unable to locate package libgconf2-4

abampalwar, 2020-09-15:
tried this 
$ sudo apt-get install libgconf-2-4
but still giving 
Package libgconf-2-4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libgconf-2-4' has no installation candidate

 $ sudo apt-get install libgconf-2-4:arm64
Then tried this but still not found any solution.

lurch, 2020-09-15:
Maybe a duplicate of one of these issues? https://github.com/balena-io/etcher/search?q=libgconf-2-4&type=issues

I suspect the Etcher devs would find it very useful if you told them exactly which OS you're running, and on which platform...

shawaj, 2020-12-15:
@abampalwar could you try `sudo apt-get --fix-broken install` 

And also could you let us know which version of Etcher you were trying to install and which OS?