balena-io/etcher #3303: Validate image with 'Flash from URL'

vpetersson, 2020-09-17:
I'm loving the new [Flash from URL](https://github.com/balena-io/etcher/pull/3071) feature. 

I have one feature request that would make this even better (IMHO): validate the downloaded file using MD5/SHA256/SHA512.

The naming convention for these checksum files is fairly standardized (i.e. just appended `foobar.zip` with the corresponding checksum of `foobar.zip.md5`).  Because a lot of failed flashed stem from corrupt downloads, this would help close that gap.

lurch, 2020-09-18:
See also the related discussion in #1113

vpetersson, 2021-04-01:
Are we the only ones who would love to see this feature? Raspberry Pi Imager does something similar by calculating the SH512 of the extracted file: https://github.com/raspberrypi/rpi-imager/blob/qml/main.qml#L409

lurch, 2021-04-01:
Raspberry Pi Imager gets that information from a [metadata file](http://downloads.raspberrypi.org/os_list_imagingutility_v2.json), and as you can see not all images include the sha256 checksum. Obviously with balenaEtcher being "more generic" there's not a similar metadata file for it to rely on.

Also, for a file called `foobar.zip` the corresponding checksum file (if there is one) could be called anything like `foobar.zip.md5`, `foobar.zip.MD5`, `foobar.md5`, `foobar.MD5`, `foobar.zip.md5sum`, `foobar.zip.MD5SUM`, `foobar.zip.md5sums`, `foobar.zip.MD5SUMS`, `foobar.md5sum`, `foobar.MD5SUM`, etc, etc, with the checksum-algorithm being MD5, SHA1, SHA256, SHA512, etc. etc. ?

vpetersson, 2021-04-01:
Those are fair points @lurch. 

However, I think there are some fairly common standards (`foobar.zip.md5`) that will emerge. For instance, Raspian uses the following:

* https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-full.zip
* https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-full.zip.sha256

For Screenly, we use append `.md5` to the file format (and adding sha256 too).

lurch, 2021-04-01:
P.S. Perhaps you should rename this issue to "Validate download with 'Flash from URL' "?