balena-io/etcher #3307: The writer process ended unexpectedly

curcuz, 2020-09-21:
- **Etcher version:** 1.5.109
- **Operating system and architecture:** Linux pop-os 5.4.0-7642-generic #46~1598628707~20.04~040157c-Ubuntu SMP Fri Aug 28 18:02:16 UTC  x86_64 x86_64 x86_64 GNU/Linux
- **Image flashed:** [raspberry pi OS Lite](https://www.raspberrypi.org/downloads/raspberry-pi-os/)
- **Do you see any meaningful error information in the DevTools?** 

```
Error: EIO: i/o error, write
code: "EUNPLUGGED"
device: "/dev/sdc"
errno: -5
name: "Error"
syscall: "write"
message: "EIO: i/o error, write"
stack: "Error: EIO: i/o error, write"
```

```
{"stack":"Error: The writer process ended unexpectedly\n    at createError (file:///tmp/.mount_balena4YyRoz/resources/app/generated/gui.js:63120:23)\n    at createUserError (file:///tmp/.mount_balena4YyRoz/resources/app/generated/gui.js:63131:16)\n    at Server.<anonymous> (file:///tmp/.mount_balena4YyRoz/resources/app/generated/gui.js:79338:28)","message":"The writer process ended unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"image":"2020-08-20-raspios-buster-armhf-lite.zip"}
```

I also tried manually unzipping the archive and feeding the image to Etcher, with the same error result

christiantreutler-avi, 2020-09-22:
Same issue. Mac version of Etcher. with this Raspi Lite OS image.

thundron, 2020-09-23:
Can you try with a different card reader/usb slot and with a different target usb/sdcard?
The `EUNPLUGGED` event is fired only when the target media is not available anymore, usually because it's been unplugged

curcuz, 2020-09-23:
> Can you try with a different card reader/usb slot and with a different target usb/sdcard?
> The `EUNPLUGGED` event is fired only when the target media is not available anymore, usually because it's been unplugged

@thundron this was a balenaFin for me

thundron, 2020-09-23:
I see, can you still try with a different target? That would help narrow down the issue in case it's just with the device. Trying with a different OS would also help in case it's an issue with the OS instead

@christiantreutler-avi same applies

christiantreutler-avi, 2020-09-25:
@thundron I tried it with a different device as well. The Raspi SD card writer works fine. For me it looks like this version of etcher has an issue.
This morning I tried another image file. Doesnt work either. I am on MacOS Mojave (latest).
If you tell me where I can find debug logs I would happily do so.

BadCarma78, 2020-09-29:
Hi I'm running MacOS High Sierra and I have the same issue but with the Ubuntu Mate 20.04.1 iso (checksum  c78cbaa40de6189e61ed5a70967d9a28148d72e9080441a55a578ee5f69d6065). I've managed to get the "regular" Ubuntu 20.04.1 on the same USB-stick. Tried another USB-stick same problem with the Official Mate iso. Found a Ubuntu mate-iso 20.04 prepared for 32-bit EFI Mac, this one also worked. So you should be able to reproduce the issue.

BadCarma78, 2020-09-29:
> Hi I'm running MacOS High Sierra and I have the same issue but with the Ubuntu Mate 20.04.1 iso (checksum c78cbaa40de6189e61ed5a70967d9a28148d72e9080441a55a578ee5f69d6065). I've managed to get the [...]

UPDATE: I managed to get it to work. I disabled McAfee Endpoint Security Threat Prevention. Unmounted the partition, not the USB-drive. Then Reflashed the USB-drive through etcher.

Ack-Levi, 2020-10-02:
I am trying it with the latest windows and am having these issues too.

wrogner, 2020-10-04:
Same issue on MacOS 10.15.7

MrPhil17, 2020-10-09:
I have the same issue on macOS Catalina 10.15.7.  If I start the app from terminal with root permission it work at first or second try.
The command is: `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`

Floppy, 2020-10-10:
Same here, macOS Catalina, Etcher 1.5.109. Output from running via sudo as @MrPhil17 said:

```
(node:82926) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/xxhash/build/Release/hash.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:82926) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/ext2fs/build/Release/bindings.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:82926) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/drivelist/build/Release/drivelist.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:82926) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/@balena.io/usb/build/Release/usb_bindings.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
```

Floppy, 2020-10-10:
I rolled back a few versions, to `1.5.106`, and had the same problem. Then I tried going *way* back, to `1.5.70`, and it works! Something's changed :)

michaelmcdonald, 2020-10-11:
> I rolled back a few versions, to `1.5.106`, and had the same problem. Then I tried going _way_ back, to `1.5.70`, and it works! Something's changed :)

Confirmed this correct my issue as well. Every OS / card / USB port I tried failed. Rolled back to 1.5.70 and issue was fixed.

Running on macOS 10.14.6

Floppy, 2020-10-12:
Good to hear! First person to bisect and find the actual version where it breaks gets this cookie -> 🍪  😁

NetForces, 2020-10-16:
Getting this same error in Windows 10 Pro build 1909 with v 1.5.109,  1.5.71 and  1.5.70.

[image]

USB drive was previously flashed with an hybrid ISO image.

Tried starting Etcher as Administrator and it does the same.

In this state Windows does not recognize the USB drive at all and you can't format it using the "Disk Management" tool

Formatted the USB drive as FAT on macOS and tried to flash with Etcher on Windows and this time it worked. Repeated the process a second time with the same key (format to FAT on macOS and flash in Windows) and it worked again.

I tried formatting the USB drive with Windows `diskpart` tool, but it fails, even as an administrator...

[image]

KatyKoenen, 2020-10-19:
Having a similar problem on a Windows 10 machine (Windows 10 Enterprise, v1903)
Using balenaEtcher v1,5,109

Trying to flash a brand new micro SD card, I get the following error:
[image]

I'm trying to flash the following Chirpstack image
[image]

MrPhil17, 2020-11-05:
Any good news?
Meanwhile I noticed that if I let it flash from URL method it work flawlessly. The issue occurs only while flashing a downloaded ISO.

rosselm, 2020-11-17:
I concur with MrPhil17 - interestingly, when Flash From File, i get certificate errors that probably can explain this bug everybody seems to have, i tried on windows, macos, same error:
❯ sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
Password:
(node:58460) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/xxhash/build/Release/hash.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:58460) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/ext2fs/build/Release/bindings.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:58460) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/drivelist/build/Release/drivelist.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(node:58460) Electron: Loading non-context-aware native module in renderer: '/Applications/balenaEtcher.app/Contents/Resources/app/generated/modules/@balena.io/usb/build/Release/usb_bindings.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
Checking for update
Generated new staging user ID: 64b49c0b-e67f-5c5b-ba38-604b07b627e7
ready-to-show: 619.027ms
Error: Error: net::ERR_CERT_INVALID
    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)
    at SimpleURLLoaderWrapper.emit (events.js:223:5)
{"stack":"Error: net::ERR_CERT_INVALID\n    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)\n    at SimpleURLLoaderWrapper.emit (events.js:223:5)","message":"net::ERR_CERT_INVALID"}
{"stack":"Error: net::ERR_CERT_INVALID\n    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)\n    at SimpleURLLoaderWrapper.emit (events.js:223:5)","message":"net::ERR_CERT_INVALID"}
Checking for update
Error: Error: net::ERR_CERT_INVALID
    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)
    at SimpleURLLoaderWrapper.emit (events.js:223:5)
{"stack":"Error: net::ERR_CERT_INVALID\n    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)\n    at SimpleURLLoaderWrapper.emit (events.js:223:5)","message":"net::ERR_CERT_INVALID"}
{"stack":"Error: net::ERR_CERT_INVALID\n    at SimpleURLLoaderWrapper.<anonymous> (electron/js2c/browser_init.js:93:6395)\n    at SimpleURLLoaderWrapper.emit (events.js:223:5)","message":"net::ERR_CERT_INVALID"}

bcomnes, 2021-01-09:
I had to restart Balena to make it work.  I think you may have to do this after you grant write permission to removable drives in Big Sur.  ymmv

MrMan999, 2021-02-05:
I had this issue too, but with another image for a NanoPi based board.  I fixed it by formatting it with FAT32 and MBR scheme.

killDevils, 2021-03-31:
Under macOS Catalina 10.15.7, with Etcher 1.5.116, fixed.

closed 2023-07-12 by dfunckt