balena-io/etcher #3309: Why aren't you using the system password dialog?

aaronraimist, 2020-09-25:
- **Etcher version:** 1.5.109
- **Operating system and architecture:** macOS 10.15.6

I just upgraded to version 1.5.109 from a version several months old and now the password dialog has changed from:

[image]

to this nonstandard password dialog:
[image]

It feels **very sketchy**. Why are you using your own password dialog?

lurch, 2020-09-25:
It's a Catalina thing - see https://github.com/balena-io/etcher/issues/3107#issuecomment-602047807

thundron, 2020-09-28:
Correct, see the other issues for more info. Closing

closed 2020-09-28 by thundron