balena-io/etcher #3310: Fails to etch Intel SSD Firmware Updater Correctly - No error, but USB drive is unusable

CvW, 2020-09-25:
Etcher version: 1.5.59
Operating system and architecture: OS X, 10.14.6 running on a MacBook Pro (2016)
Image flashed: [ISSDFUT_644_3.0.10.iso (Intel SSD Firmware Updating Tool)
](https://downloadcenter.intel.com/downloads/eula/29722/Intel-SSD-Firmware-Update-Tool?httpDown=https%3A%2F%2Fdownloadmirror.intel.com%2F29722%2Feng%2Fissdfut_3.0.10.zip)Do you see any meaningful error information in the DevTools? No, flash executes and verifies without error 

BalenaEtcher will flash this file just fine, verify it also, but the resulting ISO on a USB stick does not mount in OSX, nor will it boot on a Intel machine, such as my server. The boot process there stops with "Failed to load ldlinux.c32". Happened on multiple drives, same error message every time.

Burn the exact same ISO image via Rufus to the same USB sticks and the resulting drive is mountable in OSX  (FAT32), boots great on my server, and the SSD firmware utility works as expected.

lurch, 2020-09-25:
Possibly a MBR vs GPT thing? Or maybe the ISO is only designed to be written to a CD / DVD and _isn't_ a "hybrid ISO" that can be written to a CD **or** a USB drive? https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos
(the flashing-mode of Etcher works differently to the flashing-mode of Rufus)

thundron, 2020-09-28:
Etcher flashes images byte-to-byte, as a result if the image is not bootable by itself it probably needs either extra steps to make it bootable beforehand/afterwards, or the use of a different tool (like Rufus, as you mentioned)

closed 2020-09-28 by thundron

CvW, 2020-09-28:
> Possibly a MBR vs GPT thing? Or maybe the ISO is only designed to be written to a CD / DVD and _isn't_ a "hybrid ISO" that can be written to a CD **or** a USB drive? [...]
I doubt the issue is associated with the media itself since Rufus succeeded in making a bootable media drive on the same USB stick with the same ISO that BalenaEtcher also wrote to and successfully verified (but whose contents were not readable by either OS X or my server). It's not the hardware, it's not the ISO, it's whatever happens as part of the prep, install, and clean-up of the USB drive. 

If there are extra steps that have to be taken to make an ISO bootable, then this is a potential improvement / development opportunity for BalenaEtcher to reach functional parity with Rufus. 

As a starting point, it would be great if BalenaEtcher could flag that something went wrong - for example, flag for the user that the expected file system (in this case FAT32) is not mountable for some reason and to use another tool like Rufus instead of BalenaEtcher that does the "extra steps" (whatever they may be.

lurch, 2020-09-28:
> it's not the ISO

Sounds to me like it _is_ the ISO file. "Pure ISO" files contain an iso9660 filesystem, and are designed _only_ to be written to a CD or DVD drive, not written to a USB flash drive. "Hybrid ISO" files (see the link I posted earlier) are able to be written to a CD/DVD drive **or** a USB drive.
If the particular ISO you're using isn't bootable when written by Etcher, but is bootable when written by Rufus, then that would indicate that it's probably a "pure ISO"; which Rufus has extra logic for dealing with, but which Etcher doesn't - https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

> As a starting point, it would be great if BalenaEtcher could flag that something went wrong

I wonder why you didn't get the https://github.com/balena-io/etcher/blob/master/lib/shared/messages.ts#L112 warning from Etcher?

CvW, 2020-09-28:
Thank you for the education, my suggestion re: enhancing BalenaEtcher to be able the extra steps that Rufus does to make things bootable still stands, however. CDs are going out of fashion, flash is now far more common and it would be great if BalenaEtcher could accommodate OEMs like Intel that, for whatever reason, still use non-hybrid file system for their ISOs. 

But no, I did not get an error message. The image flashed and verified fine on two separate USB flash drives. Neither was mountable by OSX, however, and that should have been my hint that BalenaEtcher was the wrong tool for this particular ISO. Cheers and thanks, CvW

lurch, 2020-09-28:
> Neither was mountable by OSX

I don't have a Mac myself, but both pure and hybrid ISOs _may_ be non-mountable by OSX, see e.g. https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos