balena-io/etcher #3311: is owned by root and has mode 4755 open Etcher Error

auto-p, 2020-10-01:
- **Etcher version: **balenaEtcher-1.5.109-x64.AppImage**

- **Operating system and architecture:*Debian 10 bullseye GNU/Linux *

- **Do you see any meaningful error information in the DevTools?* [8615:1001/102614.244965: FATAL:setuid_ sandbox_ host.cc (158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_ balenaGkCGF0/chrome-sandbox is owned by root and has mode 4755.

/tmp/.mount_ Balenagkcgf0 / balena etcher electron: Line 10: 8615 trace and breakpoint trap "${script_ dir}"/balena-etcher- electron.bin "$@"*

It can only be opened using root

lurch, 2020-10-01:
Duplicate of #3129 ?  (although I've no idea why that issue has been closed)

See also https://github.com/balena-io/etcher/search?q=sandbox&type=issues

thundron, 2020-10-01:
Duplicate of also https://github.com/balena-io/etcher/issues/3293 which is still open and we're going to use as tracker (< reason for closing the other). Closing this one

closed 2020-10-01 by thundron